platformio/platformio-atom-ide #207: minor library issue / enhancement

rwalli, 2017-01-25:
Hallo!

On OSX PlatformIO, version 3.3.0a7 PlatformIO IDE 2.0.0-alpha.2 go to:
Home->Libraries->pick any Library->Headers->klick on any header-file->does nothing.

[image]

I dont know if this is intended or not, but it would be nice if you can view the header (or even c++) file.

Robert

ivankravets, 2017-01-25:
The idea of this functionality is to copy header file to buffer. We don't have source code of each headers. Only names.

rwalli, 2017-01-25:
ah, of course it's the clipboard icon, sorry i did not realized that!

Robert

PS:
with the header sourcecode i meant the sourcecode of the installed headers.
For now i click the url of the library and search on the website for the source if i want to look for function definition in the libraries.
I guess the right way to do this task.

ivankravets, 2017-01-25:
I've just changed a title of this issue. I propose to call it "Library API Explorer". I like this idea.