platformio/platformio-atom-ide #208: auomatic package management is invasive

smldis, 2017-01-26:
Hello, I didn't found in the main page where to ask before posting an issue. 
So sorry if this issue is a duplicate.

The problem is I have already a package manager for atom: https://github.com/dvcrn/proton/
If I try to install platformio-atom-ide it interfers with my package management in an invasive manner. How do I turn off this behaviour?
There is no configuration option for the deactivation of the platformio automatic package installer.

Thank you very much

ivankravets, 2017-01-26:
I'm not sure that this issue is related to PlatformIO. Need to ask @dvcrn to create new "platformio" layer with [these depndencies](https://github.com/platformio/platformio-atom-ide/blob/develop/lib/config.js#L93).

smldis, 2017-01-26:
ok, those dependencies should not go to a layer, for example minimap is not a real dependecy (See https://github.com/platformio/platformio-atom-ide/issues/195).

Which ones are the real dependencies if someone would like to have a layer with the same user experience of platformio IDE? Is this possible?