platformio/platformio-atom-ide #209: Support for turning on and turning off Platformio virtualenv from within atom IDE

krish2487, 2017-01-28:
Hello.
I would like to request the developers, if they can spare some time to create a Platformio specific toggle option for turning on and turning off the platformio virtualenv to allow access to the system python env.

OS : MacOS Sierra
Anaconda Version : 4.2 (Python Version 2.7)
Installation Procedure : Download the graphical installer package from the website and install.
Additional packages in atom : script

Issue : Anaconda replaces the system python with its python. When I try to run any anaconda targeted python script from with atom/platformio, due to the clean virtualenv of platformio, the script fails to run since the native python env is not loaded. hacking the init.coffee script or the platformio is not a viable option since installing the anaconda dependancies is too huge and cumbersome, and inheriting the system libraries would defeat the purpose of having a virtualenv in the first place. 

Providing a toggle for activating/deactivating the platformio virtualenv and reloading the IDE will be great help.
Thank you for the support so far !! :-)

ivankravets, 2017-02-21:
Could you try new PlatformIO IDE 2.0 Installer?

Menu: PlatformIO > Home > Try PlatformIO IDE 2.0 Preview.

Thanks!

krish2487, 2017-02-21:
I am unable to see the option @ivankravets . Should I pull the git repo and install again???
I have run the upgrade and update commands.

ivankravets, 2017-02-21:
Please try to remove `penv` folder and restart Atom. It should handle a built-in conda installation automatically.

1. Settings > Packages > filter for `platformio-ide`
2. Uninstall current GIT version
3. Install stable PIO IDE 1.7 via Settings > Install > platformio-ide
4. Switch to PIO IDE 2.0 Preview (from IDE Home screen)

closed 2017-02-21 by ivankravets

krish2487, 2017-02-21:
Works like a charm @ivankravets :-)

Thank you, I am able to compile and run the anaconda scripts as well, without changing anything!! :-D
👍

commit cecd5554 (2017-02-21): Use “conda” (if exists) to create virtual environment // Resolve #152 Resolve #174 Resolve #176 Resolve #209 Resolve #218