platformio/platformio-atom-ide #210: Uncaught SyntaxError: Unexpected end of JSON input

Roman3349, 2017-02-03:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.13.1
**Electron Version**: 1.3.13
**System**: Linux Mint
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v2.0.0-alpha.4

### Stack Trace

Uncaught SyntaxError: Unexpected end of JSON input

```
At file:///usr/share/atom/resources/app.asar/static/index.html#%7B%22initialPaths%22%3A%5B%22%2Fhome%2Froman%2Fgithub%22%2C%22%2Fhome%2Froman%2Fgithub%2Fclibdpa-mcu%22%2C%22%2Fhome%2Froman%2Fgithub%2Fiqrf-daemon%22%2C%22%2Fhome%2Froman%2Fgithub%2Fpylibiqrf%22%2C%22%2Fhome%2Froman%2Fgithub%2Frpi-gw%22%2C%22%2Fhome%2Froman%2Fgithub%2Fsouteze%2Frpi-gw%22%2C%22%2Fhome%2Froman%2Fgithub%2Fsouteze%2Fsmart-socket%22%5D%2C%22locationsToOpen%22%3A%5B%7B%22pathToOpen%22%3A%22%2Fhome%2Froman%2Fgithub%2Fsouteze%2Frpi-gw%22%2C%22initialLine%22%3Anull%2C%22initialColumn%22%3Anull%2C%22forceAddToWindow%22%3Afalse%7D%2C%7B%22pathToOpen%22%3A%22%2Fhome%2Froman%2Fgithub%2Fsouteze%2Fsmart-socket%22%2C%22initialLine%22%3Anull%2C%22initialColumn%22%3Anull%2C%22forceAddToWindow%22%3Afalse%7D%2C%7B%22pathToOpen%22%3A%22%2Fhome%2Froman%2Fgithub%2Fpylibiqrf%22%2C%22initialLine%22%3Anull%2C%22initialColumn%22%3Anull%2C%22forceAddToWindow%22%3Afalse%7D%2C%7B%22pathToOpen%22%3A%22%2Fhome%2Froman%2Fgithub%22%2C%22initialLine%22%3Anull%2C%22initialColumn%22%3Anull%2C%22forceAddToWindow%22%3Afalse%7D%2C%7B%22pathToOpen%22%3A%22%2Fhome%2Froman%2Fgithub%2Fclibdpa-mcu%22%2C%22initialLine%22%3Anull%2C%22initialColumn%22%3Anull%2C%22forceAddToWindow%22%3Afalse%7D%2C%7B%22pathToOpen%22%3A%22%2Fhome%2Froman%2Fgithub%2Fiqrf-daemon%22%2C%22initialLine%22%3Anull%2C%22initialColumn%22%3Anull%2C%22forceAddToWindow%22%3Afalse%7D%2C%7B%22pathToOpen%22%3A%22%2Fhome%2Froman%2Fgithub%2Frpi-gw%22%2C%22initialLine%22%3Anull%2C%22initialColumn%22%3Anull%2C%22forceAddToWindow%22%3Afalse%7D%5D%2C%22windowInitializationScript%22%3A%22%2Fusr%2Fshare%2Fatom%2Fresources%2Fapp.asar%2Fsrc%2Finitialize-application-window.js%22%2C%22resourcePath%22%3A%22%2Fusr%2Fshare%2Fatom%2Fresources%2Fapp.asar%22%2C%22devMode%22%3Afalse%2C%22safeMode%22%3Afalse%2C%22profileStartup%22%3Afalse%2C%22clearWindowState%22%3Afalse%2C%22env%22%3A%7B%22GNOME_KEYRING_PID%22%3A%222561%22%2C%22USER%22%3A%22roman%22%2C%22LC_TIME%22%3A%22cs_CZ.UTF-8%22%2C%22J2SDKDIR%22%3A%22%2Fusr%2Flib%2Fjvm%2Fjava-8-oracle%22%2C%22XDG_SEAT%22%3A%22seat0%22%2C%22TEXTDOMAIN%22%3A%22im-config%22%2C%22SSH_AGENT_PID%22%3A%222706%22%2C%22J2REDIR%22%3A%22%2Fusr%2Flib%2Fjvm%2Fjava-8-oracle%2Fjre%22%2C%22KIGITHUB%22%3A%22https%3A%2F%2Fgithub.com%2FKiCad%22%2C%22HOME%22%3A%22%2Fhome%2Froman%22%2C%22MDM_LANG%22%3A%22cs_CZ.UTF-8%22%2C%22XDG_SESSION_COOKIE%22%3A%2225a491dd49a7810a74415a825703c2a3-1486107643.21644-1337608256%22%2C%22DESKTOP_SESSION%22%3A%22cinnamon%22%2C%22LC_MONETARY%22%3A%22cs_CZ.UTF-8%22%2C%22GDM_XSERVER_LOCATION%22%3A%22local%22%2C%22DBUS_SESSION_BUS_ADDRESS%22%3A%22unix%3Aabstract%3D%2Ftmp%2Fdbus-sdBHiC82qI%2Cguid%3D9bb9fc290818cd6c028fc1c3589433fb%22%2C%22MDMSESSION%22%3A%22cinnamon%22%2C%22GNOME_KEYRING_CONTROL%22%3A%22%2Frun%2Fuser%2F1000%2Fkeyring-RnhBf3%22%2C%22MANDATORY_PATH%22%3A%22%2Fusr%2Fshare%2Fgconf%2Fcinnamon.mandatory.path%22%2C%22LOGNAME%22%3A%22roman%22%2C%22DEFAULTS_PATH%22%3A%22%2Fusr%2Fshare%2Fgconf%2Fcinnamon.default.path%22%2C%22USERNAME%22%3A%22roman%22%2C%22XDG_SESSION_ID%22%3A%22c1%22%2C%22WINDOWPATH%22%3A%228%22%2C%22PATH%22%3A%22%2Fusr%2Flocal%2Fsbin%3A%2Fusr%2Flocal%2Fbin%3A%2Fusr%2Fsbin%3A%2Fusr%2Fbin%3A%2Fsbin%3A%2Fbin%3A%2Fusr%2Fgames%3A%2Fusr%2Flocal%2Fgames%3A%2Fusr%2Flib%2Fjvm%2Fjava-8-oracle%2Fbin%3A%2Fusr%2Flib%2Fjvm%2Fjava-8-oracle%2Fdb%2Fbin%3A%2Fusr%2Flib%2Fjvm%2Fjava-8-oracle%2Fjre%2Fbin%22%2C%22PAPERSIZE%22%3A%22a4%22%2C%22DERBY_HOME%22%3A%22%2Fusr%2Flib%2Fjvm%2Fjava-8-oracle%2Fdb%22%2C%22XDG_RUNTIME_DIR%22%3A%22%2Frun%2Fuser%2F1000%22%2C%22LC_ADDRESS%22%3A%22cs_CZ.UTF-8%22%2C%22DISPLAY%22%3A%22%3A0%22%2C%22LANG%22%3A%22cs_CZ.UTF-8%22%2C%22LC_TELEPHONE%22%3A%22cs_CZ.UTF-8%22%2C%22XAUTHORITY%22%3A%22%2Fhome%2Froman%2F.Xauthority%22%2C%22XDG_SESSION_DESKTOP%22%3A%22cinnamon%22%2C%22SSH_AUTH_SOCK%22%3A%22%2Frun%2Fuser%2F1000%2Fkeyring-RnhBf3%2Fssh%22%2C%22SHELL%22%3A%22%2Fbin%2Fbash%22%2C%22LC_NAME%22%3A%22cs_CZ.UTF-8%22%2C%22GDMSESSION%22%3A%22cinnamon%22%2C%22LC_MEASUREMENT%22%3A%22cs_CZ.UTF-8%22%2C%22LC_IDENTIFICATION%22%3A%22cs_CZ.UTF-8%22%2C%22TEXTDOMAINDIR%22%3A%22%2Fusr%2Fshare%2Flocale%2F%22%2C%22XDG_VTNR%22%3A%228%22%2C%22PWD%22%3A%22%2Fhome%2Froman%22%2C%22JAVA_HOME%22%3A%22%2Fusr%2Flib%2Fjvm%2Fjava-8-oracle%22%2C%22KICAD_PATH%22%3A%22%2Fusr%2Fshare%2Fkicad%22%2C%22XDG_DATA_DIRS%22%3A%22%2Fusr%2Fshare%2Fcinnamon%3A%2Fusr%2Fshare%2Fgnome%3A%2Fusr%2Flocal%2Fshare%2F%3A%2Fusr%2Fshare%2F%3A%2Fusr%2Fshare%2Fmdm%2F%22%2C%22XDG_CONFIG_DIRS%22%3A%22%2Fetc%2Fxdg%2Fxdg-cinnamon%3A%2Fetc%2Fxdg%22%2C%22MDM_XSERVER_LOCATION%22%3A%22local%22%2C%22LC_NUMERIC%22%3A%22cs_CZ.UTF-8%22%2C%22LC_PAPER%22%3A%22cs_CZ.UTF-8%22%2C%22GNOME_DESKTOP_SESSION_ID%22%3A%22this-is-deprecated%22%2C%22CLUTTER_BACKEND%22%3A%22x11%22%2C%22QT_STYLE_OVERRIDE%22%3A%22gtk%22%2C%22SESSION_MANAGER%22%3A%22local%2FASUS-B85-PRO-GAMER%3A%40%2Ftmp%2F.ICE-unix%2F2641%2Cunix%2FASUS-B85-PRO-GAMER%3A%2Ftmp%2F.ICE-unix%2F2641%22%2C%22XDG_CURRENT_DESKTOP%22%3A%22X-Cinnamon%22%2C%22GPG_AGENT_INFO%22%3A%22%2Frun%2Fuser%2F1000%2Fkeyring-RnhBf3%2Fgpg%3A0%3A1%22%2C%22GJS_DEBUG_OUTPUT%22%3A%22stderr%22%2C%22GJS_DEBUG_TOPICS%22%3A%22JS%20ERROR%3BJS%20LOG%22%2C%22CINNAMON_VERSION%22%3A%222.8.8%22%2C%22GIO_LAUNCHED_DESKTOP_FILE%22%3A%22%2Fusr%2Fshare%2Fapplications%2Fatom.desktop%22%2C%22GIO_LAUNCHED_DESKTOP_FILE_PID%22%3A%227230%22%2C%22GOOGLE_API_KEY%22%3A%22AIzaSyAQfxPJiounkhOjODEO5ZieffeBv6yft2Q%22%2C%22CHROME_DESKTOP%22%3A%22Atom.desktop%22%2C%22ATOM_HOME%22%3A%22%2Fhome%2Froman%2F.atom%22%7D%2C%22appVersion%22%3A%221.13.1%22%2C%22atomHome%22%3A%22%2Fhome%2Froman%2F.atom%22%2C%22shellLoadTime%22%3A940%7D:1

SyntaxError: Unexpected end of JSON input
    at Object.parse (native)
    at /home/roman/.atom/packages/platformio-ide/lib/libraries/util.js:99:61
    at exit (/home/roman/.atom/packages/platformio-ide/lib/utils.js:158:9)
    at triggerExitCallback (/usr/share/atom/resources/app.asar/src/buffered-process.js:299:11)
    at /usr/share/atom/resources/app.asar/src/buffered-process.js:320:11
    at Socket.<anonymous> (/usr/share/atom/resources/app.asar/src/buffered-process.js:181:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```

### Commands

```
     -1:07.7.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:43 tree-view:add-folder (span.name.icon.icon-file-directory)
     -0:32 core:paste (input.hidden-input)
     -0:31.3.0 core:confirm (input.hidden-input)
```

### Config

```json
{
  "core": {
    "ignoredNames": [
      ".git",
      ".hg",
      ".svn",
      ".DS_Store",
      "._*",
      "Thumbs.db",
      ".pioenvs",
      ".piolibdeps",
      ".clang_complete",
      ".gcc-flags.json"
    ],
    "telemetryConsent": "limited"
  }
}
```

### Installed Packages

```coffee
# User
atom-beautify, v0.29.17 (inactive)
atom-material-ui, v1.3.9 (inactive)
autocomplete-clang, v0.10.0 (active)
autocomplete-cmake, v0.3.1 (active)
build, v0.67.0 (active)
build-cmake, v0.8.3 (active)
[... 112 lines cut ...]
language-javascript, v0.122.0 (active)
language-json, v0.18.3 (active)
language-less, v0.29.6 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.37.0 (active)
language-php, v0.37.3 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.1 (active)
language-ruby, v0.70.2 (active)
language-ruby-on-rails, v0.25.1 (active)
language-sass, v0.57.0 (active)
language-shellscript, v0.23.0 (active)
language-source, v0.9.0 (active)
language-sql, v0.25.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.29.1 (active)
language-toml, v0.18.1 (active)
language-xml, v0.34.12 (active)
language-yaml, v0.27.1 (active)

# Dev
No dev packages
```

ivankravets, 2017-02-03:
Which version of PIO Core do you see on home screen? Please remove from a system any duplicates of PIO Core. Open system terminal and run `pip uninstalled PlatformIO` and install IDE Core to system: Menu - PlatformIO- Install Shell Commands

Roman3349, 2017-02-04:
**PIO Core:** 3.3.0a10
**IDE**: 2.0.0-alpha.5

No change after uninstalling PIO Core and installing new PIO Core from Atom.

Roman3349, 2017-02-04:
It was caused an outdated platform espressif.

closed 2017-02-04 by Roman3349