platformio/platformio-atom-ide #211: Uncaught Error: Cannot create cache directory

graffiti4u, 2017-02-03:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.13.1
**Electron Version**: 1.3.13
**System**: Mac OS X 10.12.3
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.7.2

### Stack Trace

Uncaught Error: Cannot create cache directory

```
At /Users/graffiti/.atom/packages/platformio-ide/lib/install/command.js:229

Error: Cannot create cache directory
    at /Users/graffiti/.atom/packages/platformio-ide/lib/install/command.js:229:19
    at FSReqWrap.oncomplete (fs.js:123:15)
```

### Commands

```
     -0:01.3.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
```

### Config

```json
{
  "core": {
    "disabledPackages": [
      "wrap-guide",
      "welcome",
      "docblockr",
      "language-javascript-jsx"
    ],
    "ignoredNames": [
      ".git",
      ".hg",
      ".svn",
      ".DS_Store",
      "._*",
      "Thumbs.db",
      ".pioenvs",
      ".piolibdeps",
      ".clang_complete",
      ".gcc-flags.json"
    ],
    "telemetryConsent": "limited"
  },
  "platformio-ide": {
    "showHomeScreen": false
  }
}
```

### Installed Packages

```coffee
# User
aligner-php, v1.0.0 (active)
angularjs, v0.3.5 (inactive)
atom-autocomplete-php, v0.21.0 (active)
atom-beautify, v0.29.17 (inactive)
atom-bracket-highlight, v0.3.2 (active)
atom-html-preview, v0.1.22 (active)
atom-jade, v0.3.0 (inactive)
atom-monokai, v0.10.7 (inactive)
autocomplete-clang, v0.10.0 (active)
autocomplete-php, v0.3.7 (active)
autocomplete-python, v1.8.37 (active)
build, v0.67.0 (active)
busy, v0.7.0 (active)
color-picker, v2.2.5 (active)
convert-to-utf8, v0.2.4 (inactive)
docblockr, v0.9.1 (inactive)
emmet, v2.4.3 (active)
ex-mode, v0.13.1 (inactive)
file-icons, v2.0.14 (active)
git-plus, v7.2.0 (active)
highlight-line, v0.12.0 (active)
highlight-selected, v0.12.0 (active)
javascript-snippets, v1.2.1 (active)
language-ejs, v0.4.0 (inactive)
language-ini, v1.17.0 (inactive)
language-javascript-jsx, v0.3.7 (inactive)
linter, v1.11.21 (active)
linter-bootlint, v1.0.3 (active)
linter-csslint, v1.3.3 (active)
linter-eslint, v8.1.0 (active)
linter-gcc, v0.7.1 (active)
linter-htmlhint, v1.3.1 (active)
linter-jshint, v3.0.2 (active)
linter-php, v1.3.1 (active)
linter-phpcs, v1.5.8 (active)
linter-phpmd, v1.5.2 (active)
linter-pylint, v1.2.2 (active)
livereload, v0.4.4 (active)
minimap, v4.26.8 (active)
monokai, v0.20.0 (inactive)
my-package, v0.0.0 (inactive)
my-theme-syntax, v0.0.0 (inactive)
nucleus-dark-ui, v0.10.22 (inactive)
open-in-browser, v0.4.7 (inactive)
platformio-ide, v1.7.2 (active)
platformio-ide-terminal, v2.2.2 (active)
project-manager, v3.3.3 (active)
Remote-FTP, v0.10.5 (active)
script, v3.14.1 (inactive)
simple-drag-drop-text, v0.3.1 (active)
tool-bar, v1.0.1 (active)
vim-mode, v0.66.0 (active)
wordpress-api, v1.2.2 (inactive)
atom-dark-syntax, v0.28.0 (inactive)
atom-dark-ui, v0.53.0 (inactive)
atom-light-syntax, v0.29.0 (inactive)
atom-light-ui, v0.46.0 (inactive)
base16-tomorrow-dark-theme, v1.4.0 (inactive)
base16-tomorrow-light-theme, v1.4.0 (inactive)
one-dark-ui, v1.8.2 (active)
one-light-ui, v1.8.2 (inactive)
one-dark-syntax, v1.6.0 (active)
one-light-syntax, v1.6.0 (inactive)
solarized-dark-syntax, v1.1.1 (inactive)
solarized-light-syntax, v1.1.1 (inactive)
about, v1.7.2 (active)
archive-view, v0.62.0 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.14.1 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.33.1 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.29.0 (inactive)
autosave, v0.23.2 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.43.2 (active)
bracket-matcher, v0.85.1 (active)
command-palette, v0.39.1 (inactive)
deprecation-cop, v0.55.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.40.0 (active)
find-and-replace, v0.204.5 (inactive)
fuzzy-finder, v1.4.0 (active)
git-diff, v1.2.0 (active)
go-to-line, v0.31.2 (inactive)
grammar-selector, v0.48.2 (active)
image-view, v0.60.0 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.1 (active)
link, v0.31.2 (inactive)
markdown-preview, v0.159.3 (active)
metrics, v1.1.3 (active)
notifications, v0.65.1 (active)
open-on-github, v1.2.1 (inactive)
package-generator, v1.0.2 (inactive)
settings-view, v0.244.0 (active)
snippets, v1.0.4 (active)
spell-check, v0.68.5 (active)
status-bar, v1.6.0 (active)
styleguide, v0.48.0 (active)
symbols-view, v0.113.1 (inactive)
tabs, v0.103.1 (active)
timecop, v0.33.2 (active)
tree-view, v0.211.1 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.35.1 (inactive)
whitespace, v0.35.0 (active)
wrap-guide, v0.39.0 (inactive)
language-c, v0.54.0 (active)
language-clojure, v0.22.1 (active)
language-coffee-script, v0.48.1 (active)
language-csharp, v0.13.0 (active)
language-css, v0.40.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.43.0 (active)
language-html, v0.47.1 (active)
language-hyperlink, v0.16.1 (active)
language-java, v0.24.0 (active)
language-javascript, v0.122.0 (active)
language-json, v0.18.3 (active)
language-less, v0.29.6 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.37.0 (active)
language-php, v0.37.3 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.1 (active)
language-ruby, v0.70.2 (active)
language-ruby-on-rails, v0.25.1 (active)
language-sass, v0.57.0 (active)
language-shellscript, v0.23.0 (active)
language-source, v0.9.0 (active)
language-sql, v0.25.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.29.1 (active)
language-toml, v0.18.1 (active)
language-xml, v0.34.12 (active)
language-yaml, v0.27.1 (active)

# Dev
No dev packages
```

ivankravets, 2017-02-04:
Could you remove `Users/graffiti/.atom/packages/platformio-ide` folder and install IDE again using Settings ⚙️ > Install > platformio-ide ?

ivankravets, 2017-02-21:
Could you try new PlatformIO IDE 2.0 Installer?

Menu: PlatformIO > Home > Try PlatformIO IDE 2.0 Preview.

Thanks!

closed 2017-02-21 by ivankravets