platformio/platformio-atom-ide #213: When a filesystem path is passed in within lib_deps in platformio.ini, the index rebuilding loops

jasiek, 2017-02-07:
How to reproduce?

In your platformio.ini file, add a reference to a dependent library, that resides elsewhere on the filesystem, for example:

```
[env:nodemcuv2]
platform = espressif8266
framework = arduino
board = nodemcuv2
lib_deps =
  ArduinoJson
  DHT sensor library
  Adafruit Unified Sensor
  Adafruit BMP085 Library
  https://github.com/jasiek/Adafruit_HTU21DF_Library.git
  /Users/jps/github/metering
```

After saving that file the IDE appears to constantly re-index the project, alternating between the two dialogs attached below.
[image]

ivankravets, 2017-02-07:
Temporary solution is to move `/Users/jps/github/metering` to 
```
lib_extra_dirs = /Users/jps/github
```

ivankravets, 2017-02-21:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/179

closed 2017-02-21 by ivankravets