platformio/platformio-atom-ide #214: PlatformIO menu options not working after install

tomneutens, 2017-02-09:
Hi,

I installed the platformio atom editor on Ubuntu 16.04 LTS and have an issue when I try to use menu options like "Initialize or update PlatformIO Project", "build", "run other target", ... None of the menu options seem to do anything. The strange thing is that this doesn't happen the first time i open the ide then the menu options do work. However the second time i start the IDE they stop working. Also after a while when I try to select the menu options the full editor crashes. 

Does anyone have a solution for this problem?

Regards,
Tom

ivankravets, 2017-02-12:
Do you use the latest Atom editor? See http://Atom.io

ivankravets, 2017-02-21:
Could you try new PlatformIO IDE 2.0 Installer?

Menu: PlatformIO > Home > Try PlatformIO IDE 2.0 Preview.

Thanks!

ryanbbernstein, 2017-04-01:
I'm having the same issue. No error messages just nothing happens.
I can't open the platformio > home.

mogplus8, 2017-07-28:
I'm having the same problem. Most items under the Platformio menu item don't do anything. Some do. Open project folder, run other target, toggle build panel, terminal, rebuild c/c++ project index,  and update atom packages all work, but none of the other items do anything. Not even help or about. I have also noticed that when I start Platformio it tries to download and install the project examples, then asks to restart. If I hit restart it restarts and downloads the examples again, and asks to restart again. This is an endless loop. If I close the window (that asks for the restart) without restarting I can use PIO, but most of the PlatformIO menu items don't work as noted above. Nor do the corresponding items in the toolbar on the left hand side. 
I don't know if it's related but I've had some issues with the permissions on the .config folder in my home folder (I'm using Linux Mint 18.2) which seems to stop things from working sometimes.
I have also verified that PIO works fine in another linux partition (running Linux Mint Debian Edition) from a clean install of Atom and PlatformIO. All seems to be well there, so it must be some problem with the setup in Mint. 
I tried deleting the .atom folder in my home directory. I found that I had to install Platformio again, (using the package install in Atom), but found that the same problems persisted. 
It seems to me the best option would be to completely uninstall Platformio and start again. I've tried this a couple of times but the problems persist. There must be some more folders or files that are not deleted by uninstalling, but I don't know what they are. Or some other setting hidden in Mint somewhere that is stopping things from working. 
All advice gratefully received.
Ian

mogplus8, 2017-07-28:
I'm having the same problem. Most items under the Platformio menu item don't do anything. Some do. Open project folder, run other target, toggle build panel, terminal, rebuild c/c++ project index,  and update atom packages all work, but none of the other items do anything. Not even help or about. I have also noticed that when I start Platformio it tries to download and install the project examples, then asks to restart. If I hit restart it restarts and downloads the examples again, and asks to restart again. This is an endless loop. If I close the window (that asks for the restart) without restarting I can use PIO, but most of the PlatformIO menu items don't work as noted above. Nor do the corresponding items in the toolbar on the left hand side. 
I don't know if it's related but I've had some issues with the permissions on the .config folder in my home folder (I'm using Linux Mint 18.2) which seems to stop things from working sometimes.
I have also verified that PIO works fine in another linux partition (running Linux Mint Debian Edition) from a clean install of Atom and PlatformIO. All seems to be well there, except I can't get the menu bar to display, so it must be some problem with the setup in Mint. 
I tried deleting the .atom folder in my home directory. I found that I had to install Platformio again, (using the package install in Atom), but found that the same problems persisted. 
It seems to me the best option would be to completely uninstall Platformio and start again. I've tried this a couple of times but the problems persist. There must be some more folders or files that are not deleted by uninstalling, but I don't know what they are. Or some other setting hidden in Mint somewhere that is stopping things from working. 
All advice gratefully received.
Ian

mogplus8, 2017-07-29:
Well, I thought I'd solved the problem, but alas, no. I tried Visual Studio Code, but found that it would not even start. I see the tumbling cursor for a few seconds, and then, nothing. Thinking it must be something to do with the .config file, I checked it again. I found that the sub folders "atom" and "code" (related to VS code) were owned by "root", not by my userid. I changed the ownership and had high hopes of everything being okay, but the problems are still there. I'm going to try a new userid now to see if that changes anything.
:-(

mogplus8, 2017-07-29:
Tried creating a new userid, opened up VS code and everything worked! So did Platformio!! Double checked the permissions on the .config folder for my new userid, and found that the userid owned it, and the subfolders for "atom" and "code". Checked them for my older userid and found that .config was still owned by root. Changed that to be owned by me. Tried again. Still didn't work!!! Triple checked the permissions. Found that on the new userid my userid had read and write permissions but access was denied (none) for all other groups and users. On my old userid other groups and users had read access. Changed that to none, and VOILA! Everything works. Problem solved.

:-)

ivankravets, 2017-11-06:
Please upgrade to 2.0.0-rc.4

closed 2017-11-06 by ivankravets