platformio/platformio-atom-ide #218: Failed to retrieve PIO Core version

kriss2013, 2017-02-13:
Der Befehl "platformio" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

aufdenkampe, 2017-02-13:
I also get a "Core: Failed to retrieve" error when I launch.

This is after having numerous installation issues, that I mentioned here: https://github.com/platformio/platformio-core/issues/851#issuecomment-278357855

I did a full reinstall and finally got the install to complete by following the advice on how to link the env installed by conda, given here: platformio/platformio-atom-ide#152 (comment)

However, this again led to a "Core: Failed to retrieve" error, which I'm following up on in this issue.

I'm running MacOS 10.12.3, Python 2.7.13 |Anaconda 4.3.0 (x86_64)
On this machine, I installed Anaconda Python for all users, so that my Python path is here: //anaconda/bin/

On my home Mac this installed fine. It's the same OS & Python, but I installed just for me. I'm concerned that there is a path or permissions issue, but I can't figure it out.
Any ideas?

ivankravets, 2017-02-21:
Could you try new PlatformIO IDE 2.0 Installer?

Menu: PlatformIO > Home > Try PlatformIO IDE 2.0 Preview.

Thanks!

aufdenkampe, 2017-02-21:
I just did, and it all looked good until retrieving core. I got this error.

```
nstallationManager
Virtualenv: Traceback (most recent call last):
  File "/var/folders/tx/14mbtmrs0rq2b6mkfxdh54r00000gp/T/tmp-394167quMDHzbQT0G/virtualenv-14.0.6/virtualenv.py", line 29, in <module>
    import tempfile
  File "//anaconda/lib/python2.7/tempfile.py", line 32, in <module>
    import io as _io
  File "//anaconda/lib/python2.7/io.py", line 51, in <module>
    import _io
ImportError: dlopen(/Users/aaufdenkampe/.atom/packages/platformio-ide/penv/lib/python2.7/lib-dynload/_io.so, 2): Symbol not found: __PyCodecInfo_GetIncrementalDecoder
  Referenced from: /Users/aaufdenkampe/.atom/packages/platformio-ide/penv/lib/python2.7/lib-dynload/_io.so
  Expected in: flat namespace
 in /Users/aaufdenkampe/.atom/packages/platformio-ide/penv/lib/python2.7/lib-dynload/_io.so
Report a problem
```

ivankravets, 2017-02-21:
@aufdenkampe do you have `virtualenv` shipped with Anaconda?

Please open system terminal and type `virtualenv --version`.

aufdenkampe, 2017-02-21:
I don't think  `virtualenv` ships with Anaconda. So I tried `pip install virtualenv` as I described here https://github.com/platformio/platformio-core/issues/851#issuecomment-278357855. 

As I described in that issue comment, that didn't work, so I uninstalled PlatformIO, I did a full reinstall and finally got the install to complete by following the advice on how to link the env installed by conda, given here: https://github.com/platformio/platformio-atom-ide/issues/152#issuecomment-239866260.

I'm not sure if the `conda` installed `env` is the same as `virtualenv`, but it seemed to work for the @brainstorm in that last issue.
When I list my conda environments, I get this:
```
$ conda-env list
# conda environments:
#
platformio               //anaconda/envs/platformio
root                  *  //anaconda
```

ivankravets, 2017-02-21:
I got you. I see the workaround posted by @brainstorm 

I need more info on that:

1. Is possible to create a virtual environment in own location?
2. Could you try 
```
rm -rf ~/.atom/packages/platformio-ide/penv
conda create -p ~/.atom/packages/platformio-ide/penv python=2.7
```

See docs https://conda.io/docs/commands/conda-create.html

Need to find single command how to create virtual env with conda in custom folder.

aufdenkampe, 2017-02-21:
I tried running those two lines in the Terminal, and they seemed to run. I got this response:
```
# To activate this environment, use:
# > source activate /Users/aaufdenkampe/.atom/packages/platformio-ide/penv
#
# To deactivate this environment, use:
# > source deactivate /Users/aaufdenkampe/.atom/packages/platformio-ide/penv
```
So I ran `source activate /Users/aaufdenkampe/.atom/packages/platformio-ide/penv`
but when I reran `conda-env list`, I didn't see the new environment.

When I reopened Atom, I got this error:
```
InstallationManager
Virtualenv: Error: spawn /Users/aaufdenkampe/.atom/packages/platformio-ide/penv/bin/python ENOENT
```

Also, it's worth noting that although PlatformIO has a menu list in Atom, I can't open the "PlatformIO Home". Nothing happens, so I can't see the "Core: Failed to retrieve" error that started this issue.

ivankravets, 2017-02-21:
@aufdenkampe OK, please sorry. I've just installed Miniconda and will update current installer.

I'll report here when it will be ready for testing.

closed 2017-02-21 by ivankravets

ivankravets, 2017-02-21:
@aufdenkampe please try to remove `penv` folder and restart Atom. It should handle a built-in conda installation automatically

aufdenkampe, 2017-02-21:
Wow! Thank you @ivankravets . That did it. I actually had to completely uninstall PlatformIO-IDE, reinstall the package then try out 2.0-alpha. All working now. Thanks! I can't wait to get started on using this for our Arduino-compatible EnviroDIY Mayfly data logger project, at https://github.com/EnviroDIY.

ivankravets, 2017-02-21:
@aufdenkampe thanks a lot for the report!

red-ibanez, 2019-12-29:
Thank you @ivankravets.   I had a weird problem in which platformio seemed to be working (listing available platforms would retrieve all of them, all other commands through the CLI and IDE would also work), but I couldn't create a project or install any of the platforms.  It would respond with "Detected unknown package"  I tried everything and thought it was a python/platformio discrepancy.  It was until I moved the 'penv' folder out of the way and this folder was regenerated again, then voila.. 

For the records:
pip freeze
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
bottle==0.12.18
certifi==2019.11.28
chardet==3.0.4
Click==7.0
colorama==0.4.3
idna==2.8
marshmallow==2.20.5
platformio==4.1.0
pyelftools==0.26
pyserial==3.4
requests==2.22.0
semantic-version==2.8.4
tabulate==0.8.6
urllib3==1.25.7

I came here, after experiencing:
https://github.com/platformio/platformio-core/issues/3116?utm_source=platformio&utm_medium=piohome

commit cecd5554 (2017-02-21): Use “conda” (if exists) to create virtual environment // Resolve #152 Resolve #174 Resolve #176 Resolve #209 Resolve #218