platformio/platformio-atom-ide #220: IDE for OSX is damaged

mdanetzky, 2017-02-15:
IDE downloaded from:
http://platformio.org/get-started/ide?dl=platformio-atom-mac.zip
opens with message:
“Atom.app” is damaged and can’t be opened. You should move it to the Trash.
Application downloaded from atom.io opens without any problems.

mdanetzky, 2017-02-15:
Temporary solution is to reset the com.apple.quarantine flag.
xattr -d com.apple.quarantine Atom.app

ivankravets, 2017-02-23:
Temporary solution is to use GitHub's official Atom and manually install `platformio-ide` packages. See
* http://docs.platformio.org/en/latest/ide/atom.html#ide-installation

ivankravets, 2017-02-25:
We have updated all our pages and stopped publishing own Atom bundle.
We added instruction how to install official atom and our installer.

http://platformio.org/platformio-ide

closed 2017-02-25 by ivankravets