platformio/platformio-atom-ide #226: Installing PlatformIO separate from Atom

JamesHagerman, 2017-02-19:
Hi there,

I'm not sure if this is explained anywhere but is there a way to get PlatformIO to be installed separately from Atom under Linux?

I need to use Atom for more than just PlatformIO and it seems to completely take over Atom as it's currently set up.

Any help would be great!

orgkhnargh, 2017-02-19:
Hi @JamesHagerman,

Atom uses a directory that is specified in the `$ATOM_HOME` environment variable to store settings, packages, etc. You can have separate Atom homes for platformio and for other purposes. By default you'll have `$ATOM_HOME` value set to `/home/$USER/.atom`. To use a different directory, start Atom with a different `$ATOM_HOME` value. Example: ```env ATOM_HOME=/home/$USER/.atom-platformio atom```.

JamesHagerman, 2017-02-19:
@orgkhnargh That helps a lot! I'll try some things along that vein...

closed 2017-02-20 by JamesHagerman