platformio/platformio-atom-ide #229: InstallationManager

mdanetzky, 2017-02-22:
Error: Error: ENOTEMPTY: directory not empty, rmdir '/Users/matthias/.atom/packages/platformio-ide/project-examples'

Error was thrown during update from IDE 2.0.0-alpha.5 to IDE 2.0.0-alpha.6

System is OSX Sierra

ls -1 /Users/matthias/.atom/packages/platformio-ide/project-examples
LICENSE
atmelavr-and-arduino
atmelsam-and-arduino
desktop
espressif
ide
intel_arc32
lattice_ice40
library-manager
mbed
microchippic32
nordicnrf51
platformio-examples.png
pumbaa
raspberrypi
simba
stm32
teensy
timsp430
titiva
unit-testing
wiring-blink

ivankravets, 2017-02-22:
Thanks! Try 
```
rm -rf /Users/matthias/.atom/packages/platformio-ide/project-examples
```
and restart Atom.

closed 2017-02-22 by ivankravets

mdanetzky, 2017-02-22:
Works

commit 23b79d2f (2017-02-22): Catch removeSync errors // Resolve #229