platformio/platformio-atom-ide #230: InstallationManager

mountaindude, 2017-02-22:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/b1/d_tydw054s13bhlq4cwn99hh0000gn/T/tmp-83814XxPqm8Qq93xs/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/b1/d_tydw054s13bhlq4cwn99hh0000gn/T/tmp-83814XxPqm8Qq93xs/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/b1/d_tydw054s13bhlq4cwn99hh0000gn/T/tmp-83814XxPqm8Qq93xs/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/goran/.atom/packages/platformio-ide/penv/bin/../lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/Users/goran/.atom/packages/platformio-ide/penv/bin/../lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-02-22:
Please provide next information:

1. macOS version
2. Python version: `$ python --version`
3. Python location: `$ which python`

mountaindude, 2017-02-22:
macOS Sierra 10.12.1
python 2.7.13
/usr/local/bin/python

After a restart of Atom all seems to be ok though - maybe it was just a installation issue that sorted itself...

ivankravets, 2017-02-22:
Please reopen if you will see this issue in future.

closed 2017-02-22 by ivankravets