platformio/platformio-atom-ide #233: InstallationManager

hodorii, 2017-02-23:
Virtualenv Download: Traceback (most recent call last):
  File "/tmp/tmp-11105IVQTq0Cf3w30/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/tmp/tmp-11105IVQTq0Cf3w30/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/tmp/tmp-11105IVQTq0Cf3w30/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/tmp/tmp-11105IVQTq0Cf3w30/virtualenv-14.0.6/virtualenv.py", line 1197, in install_python

    shutil.copyfile(executable, py_executable)
  File "/usr/lib/python2.7/shutil.py", line 83, in copyfile

    with open(dst, 'wb') as fdst:
IOError: [Errno 26] Text file busy: '/home/hs/.atom/packages/platformio-ide/penv/bin/python2.7'

hodorii, 2017-02-23:
preview upgraded. and first call atom.

ivankravets, 2017-02-23:
Could you reproduce it again?
```
apm uninstall platformio-ide
apm install --production platformio/platformio-atom-ide
```

asteriskie, 2017-02-23:
*I* was able to reproduce with the commands you have provided.

ivankravets, 2017-02-23:
What is your OS?

asteriskie, 2017-02-23:
Sorry, this is a different issue than the OP, didn't work for the: TypeError: ConfigurationStage.forEach is not a function issue.  Running under Windows if it is related.

asteriskie, 2017-02-23:
...but it did work this time... which I am quite happy about.

ivankravets, 2017-02-23:
@asteriskie ": TypeError: ConfigurationStage.forEac" was fixed before. Do you have any problems now?

hodorii, 2017-02-24:
OS is ubuntu 16.10 desktop.

hodorii, 2017-02-24:
solved. thank you.

2017-02-24 9:18 GMT+09:00 hodorii <hodorii@gmail.com>:

> OS is ubuntu 16.10 desktop.
>
>
>> What is your OS?
>>
>>
>
>
>
> from HoSeong(hodorii@gmail.com)

closed 2017-02-24 by ivankravets