platformio/platformio-atom-ide #238: Uncaught Error: Python 2.7 could not be found.

minjurul, 2017-02-23:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.14.3 ia32
**Electron**: 1.3.13
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 1.7.2

### Stack Trace

Uncaught Error: Python 2.7 could not be found.

```
At C:\Users\Nisga\.atom\packages\platformio-ide\lib\utils.js:78

Error: Python 2.7 could not be found.
    at getPythonExecutable (/packages/platformio-ide/lib/utils.js:78:13)
    at Object.installCommands (/packages/platformio-ide/lib/maintenance.js:160:9)
    at /packages/platformio-ide/lib/main.js:90:72)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/app.asar/src/atom-environment.js:1164:28)
    at EventEmitter.outerCallback (/app.asar/src/application-delegate.js:320:25)
    at emitThree (events.js:116:13)
    at EventEmitter.emit (events.js:194:7)
```

### Commands

```
     -2:19.3.0 platformio-ide:import-arduino-ide-project (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  3x -0:26.5.0 core:save (input.hidden-input)
     -0:14.5.0 platformio-ide:maintenance.install-commands (input.hidden-input)
```

### Non-Core Packages

```
atom-beautify 0.29.17 
atom-runner 2.7.1 
atom-terminal 0.8.0 
autoclose-html 0.23.0 
autocomplete-python 1.8.57 
open-html-in-browser 0.1.0 
platformio-ide 1.7.2 
```

ivankravets, 2017-02-23:
Please open system `cmd` Terminal and run
```
apm uninstall platformio-ide
apm install --production platformio/platformio-atom-ide
```

ivankravets, 2017-02-25:
Please reopen if you still need a help

closed 2017-02-25 by ivankravets