platformio/platformio-atom-ide #243: Error Installing Package from Atom and Terminal

KatieAJenkins, 2017-02-28:
Tried installing package through Atom as well as using the terminal command: 
apm install --production platformio/platformio-atom-ide

I'm using a Mac.

Here is my error:
gyp info it worked if it ends with ok
gyp info using node-gyp@3.4.0
gyp info using node@4.4.5 | darwin | x64
gyp http GET https://atom.io/download/electron/v1.3.13/iojs-v1.3.13.tar.gz
gyp WARN install got an error, rolling back install
gyp ERR! install error 
gyp ERR! stack Error: connect ECONNREFUSED 54.231.120.179:443
gyp ERR! stack     at Object.exports._errnoException (util.js:870:11)
gyp ERR! stack     at exports._exceptionWithHostPort (util.js:893:20)
gyp ERR! stack     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1061:14)
gyp ERR! System Darwin 16.4.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/node-gyp/bin/node-gyp.js" "install" "--runtime=electron" "--target=1.3.13" "--dist-url=https://atom.io/download/electron" "--arch=x64" "--ensure"
gyp ERR! cwd /Users/kind2karma/.atom
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 

Thank you!

ivankravets, 2017-02-28:
Somethin is wrong with your internet connection. In any case, try this:

1. Use the latest (>1.14) Atom from http://atom.io
2. Open Atom > Settings ⚙️ > Install > `platformio-ide`
3. Restart Atom, you should see "Switch to PIO IDE 2.0" in PIO Home.

ivankravets, 2017-03-25:
Please reopen if you still need help

closed 2017-03-25 by ivankravets

charlesmade, 2017-05-15:
It help， thanks much