platformio/platformio-atom-ide #244: Stacktrace during 'Rebuild C/C++ Project Index'

PhracturedBlue, 2017-03-01:
When attempting to build espurna on Windows 10, I get the following stacktrace.    This happens within a few seconds of opening a new project (without hitting the 'build' button).  If I try to hit the 'build' button, I get a constant stream of 'Waiting to rebuild Project Index' and then a stream of the same type of stacktrace.  I will end up with hundreds of python processes filling my machine which do not go away once Atom is killed.

If I build using 'pio' from a CMD.exe shell it works properly, (and I no longer have this issue from the gui...rebuilding works fine) so this seems to be related only to the GUI interaction.  It is reproducible even if I remove the espurna git repo and re-fetch it

I am using the latest PatformIO and Atom versions.
```
platformio-ide: 1.7.2
platformio-core: 3.3.0a12
espurna: https://bitbucket.org/xoseperez/espurna.git
```
```
Cloning into 'C:\Users\hass\Documents\Arduino\espurna\code\.piolibdeps\installing-os53hx-package'...
Cloning into 'C:\Users\hass\Documents\Arduino\espurna\code\.piolibdeps\installing-qm6xgl-package'...
Cloning into 'C:\Users\hass\Documents\Arduino\espurna\code\.piolibdeps\installing-ob09to-package'...
Cloning into 'C:\Users\hass\Documents\Arduino\espurna\code\.piolibdeps\installing-dlx6uq-package'...
Error: Traceback (most recent call last):

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\__main__.py", line 106, in main

    cli(None, None, None)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 700, in __call__

    return self.main(*args, **kwargs)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 680, in main

    rv = self.invoke(ctx)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 1027, in invoke

    return _process_result(sub_ctx.command.invoke(sub_ctx))

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 873, in invoke

    return ctx.invoke(self.callback, **ctx.params)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 508, in invoke

    return callback(*args, **kwargs)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\decorators.py", line 16, in new_func

    return f(get_current_context(), *args, **kwargs)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\run.py", line 110, in cli

    results.append((envname, ep.process()))

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\run.py", line 174, in process

    result = self._run()

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\run.py", line 260, in _run

    ], self.verbose)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\run.py", line 281, in _autoinstall_libdeps

    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 508, in invoke

    return callback(*args, **kwargs)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\click\decorators.py", line 26, in new_func

    return f(get_current_context().obj, *args, **kwargs)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\lib.py", line 97, in lib_install

    lm.install(library, silent=silent, interactive=interactive)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\lib.py", line 277, in install

    if not _url else name, _requirements, silent, trigger_event)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\package.py", line 554, in install

    pkg_dir = self._install_from_piorepo(name, requirements)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\lib.py", line 256, in _install_from_piorepo

    if app.get_setting("enable_ssl") else dl_data['url'], requirements)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\package.py", line 314, in _install_from_url

    util.rmtree_(tmp_dir)

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\util.py", line 570, in rmtree_

    return rmtree(path, onerror=_onerror)

  File "C:\Program Files\Python2.7\Lib\shutil.py", line 256, in rmtree

    onerror(os.rmdir, path, sys.exc_info())

  File "c:\users\hass\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\util.py", line 568, in _onerror

    os.remove(name)

WindowsError: [Error 5] Access is denied: 'C:\\Users\\hass\\Documents\\Arduino\\espurna\\code\\.piolibdeps\\installing-eyhmwx-package'
```

PhracturedBlue, 2017-03-01:
Also, the build works fine through the IDE if I disable the 'Automatically rebuild C/C++ Project Index' option (which is what I've done for now).

ivankravets, 2017-03-01:
Is this issue a duplicate of https://github.com/platformio/platformio-atom-ide/issues/179 ?

closed 2017-03-09 by ivankravets