platformio/platformio-atom-ide #245: Exception in thread rx (Serial broken after some time)

GillesC, 2017-03-01:
After some time (~minutes) I'm unable to read my atmegap1284p using the Serial Monitor.
I'm getting the following error and Traceback:
```
Exception in thread rx:
Traceback (most recent call last):
  File "C:\Python27\Lib\threading.py", line 801, in __bootstrap_inner

    self.run()
  File "C:\Python27\Lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "c:\users\gillescallebautpc\.atom\packages\platformio-ide\penv\lib\site-packages\serial\tools\miniterm.py", line 452, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "c:\users\gillescallebautpc\.atom\packages\platformio-ide\penv\lib\site-packages\serial\serialwin32.py", line 257, in in_waiting
    raise SerialException('call to ClearCommError failed')
SerialException: call to ClearCommError failed
```
The initiated command was (from Serial Monitor): `pio -f -c atom serialports monitor --port COM10`

Host OS: Windows 10
Environment: mightycore1284

Thanks,
Gilles Callebaut

ivankravets, 2017-03-01:
Could you reconnect a board?

GillesC, 2017-03-01:
I was able to reconnect the board.
Afterwhich the same error occured.

ivankravets, 2017-03-25:
Please report your issue to https://github.com/pyserial/pyserial/issues

closed 2017-03-25 by ivankravets