platformio/platformio-atom-ide #246: Platform IO invokes multiple instances of Python

mrkaroshi, 2017-03-02:
Since using Platform IO I've noticed it's invoking multiple instances of Python 2.7 on my Mac (OSX 10.11.6, IDE 2.0.0-alpha.7, Core 3.3.0a12), is this normal? Do you think this is an issue with the IDE or the core? It maxes out my CPU after a few hours use.
[image]

ivankravets, 2017-03-02:
Thanks a lot for your report. That is not normal.

1. Please close Atom
2. Check that all `python2.7` process are halted
3. Start Atom and try to find a case where multiple `python` process are starting.

Thanks!

ivankravets, 2017-03-02:
Also, `View > Developer > Toggle Developer Tools > Console` should help us.

mrkaroshi, 2017-03-02:
Okay will do, will test over the next few hours.

mrkaroshi, 2017-03-06:
Still trying to replicate the issue, it hasn't cropped up for some reason.

ivankravets, 2017-03-08:
Please reopen this issue if you will have any problems again.

closed 2017-03-08 by ivankravets

mrkaroshi, 2017-03-13:
I think I've been able to replicate the issue, and I'm almost 100% certain the serial monitor is causing it. If I program my Teensy and let the serial monitor open I will view my data over the space of a few minutes, then close it using the shortcut 'Ctrl+C', and do this repeatedly throughout a working session. I think, for whatever reason, PlatformIO may be invoking the multiple instances to handle serial communication, then not clearing them when the communication is over. Can anyone replicate this?

chris-gunawardena, 2017-07-01:
I get this every time I open the serial monitor.
I'm using the visual studio plugin.
[image]

ivankravets, 2017-07-01:
http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

chris-gunawardena, 2017-07-01:
Thanks @ivankravets when I uninstalled Atom the issue got fixed.