platformio/platformio-atom-ide #247: Only load IDE if platformio.ini file is present

webrgp, 2017-03-02:
I wonder if there is a way to toggle the the full IDE only when platformio.ini file is present. I use atom for other projects, like webapps, and I find myself disabling/enabling the extension every time I switch.

Thanks!

ivankravets, 2017-03-02:
What do you like to disable? Toolbar ?

webrgp, 2017-03-03:
I would like to not load the extension at all, unless platformio.ini is present @ the root of the project folder.

ivankravets, 2017-03-03:
We can create separate package aka `platformio-ide-helper` and disable/enable the main package automatically. Would you like to live with these lags?

I can't imagine how does awful it will work.

webrgp, 2017-03-04:
Yes! That package can contain "Create New Platformio Project", that would create the project structure with the platformio.ino file in it. That would make atom much lighter.

Other settings for this separated package, and I am dreaming here now, would be disabling other packages that has nothing todo with Platformio (like SASS pre-processor for example).

ivankravets, 2017-03-04:
Ok, I'll have a look on this issue after 2.0 release. Thanks.

webrgp, 2017-03-04:
If you need testing on this, let me know 👍

rsispal, 2017-09-01:
Any updates on this? Getting tired of manually switching on/off the IDE

Petfrogg, 2018-01-13:
Progress on this?
Something like:
`
File -> New window [Atom IDE]
File -> New window [Platform IDE]
`
This would open up for more IDE variants, dedicated loading etc. More like opening Libre office and choosing between Writer or Calc.

nickdaria, 2018-07-16:
Bump because a configuration allowing for PlatformIO to be initialized/closed by a helper would be very helpful. Tired of having to toggle it off for unrelated projects.

ivankravets, 2018-07-17:
There are some technical issues to get this working. Nevertheless, we did this work for PlatformIO IDE for VSCode. Please try http://docs.platformio.org/en/latest/ide/vscode.html

karlbeecken, 2018-11-29:
I think that you should implement it soon, because some people are annoyed by the overloaded IDE, this speaks against the main-thoughts of the "text-editor for the 21st century" clean but mighty

sbrl, 2019-05-26:
Hello!

I'm a potential PlatformIO user, but after trying it once a year or 2 ago, I haven't installed it since because it does completely take over Atom. I use Atom to program in a diverse range of languages, and I'd really appreciate it if PlatformIO would just _go away_ when I'm working on other projects.

In fact, I'd like it to be less 'in my face' in general. I _hate_ the toolbar it adds, and I _really don't like_ the terminal package it forces on me with the extra double-toolbar at the bottom of the screen.

I'm looking for quiet integration with my existing workflow - not something that decides how I'm going to work. If I wanted something that decides my workflow for me, I'd be using venerable _Arduino IDE_ (which I'm unfortunately still having to use because I haven't found anything else that actually works :-/).

JaccoVeldscholten, 2020-12-14:
Put this in. Idiots
Bump.