platformio/platformio-atom-ide #249: Failed to load the platformio-ide package

manoj153, 2017-03-04:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.14.4 x64
**Electron**: 1.3.13
**OS**: Linux Mint
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 1.7.2

### Stack Trace

Failed to load the platformio-ide package

```
At Refused to evaluate a string as JavaScript because 'unsafe-eval' is not an allowed source of script in the following Content Security Policy directive: "script-src 'self'".

EvalError: Refused to evaluate a string as JavaScript because 'unsafe-eval' is not an allowed source of script in the following Content Security Policy directive: "script-src 'self'".

    at Ajv.localCompile (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:139:26)
    at Ajv.resolve (/packages/platformio-ide/node_modules/ajv/lib/compile/resolve.js:53:19)
    at Object.resolveRef (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:210:21)
    at /packages/platformio-ide/node_modules/ajv/lib/dotjs/ref.js:22:22)
    at /packages/platformio-ide/node_modules/ajv/lib/dotjs/validate.js:156:37)
    at /packages/platformio-ide/node_modules/ajv/lib/dotjs/properties.js:199:26)
    at generate_validate (/packages/platformio-ide/node_modules/ajv/lib/dotjs/validate.js:229:37)
    at localCompile (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:99:22)
    at Ajv.compile (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:67:13)
    at _compile (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:328:29)
    at getSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:192:51)
    at validate (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:91:11)
    at validateSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:162:19)
    at _addSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:288:7)
    at addSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:133:26)
    at addInitialSchemas (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:396:39)
    at /packages/platformio-ide/node_modules/ajv/lib/ajv.js:78:3)
    at /packages/platformio-ide/node_modules/har-validator/lib/node4/promise.js:43:11)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/platformio-ide/node_modules/request/lib/har.js:5:16)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
```

### Commands

```
     -2:32.5.0 core:paste (input.hidden-input)
     -2:29.6.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
platformio-ide 1.7.2 
platformio-ide-terminal 2.4.0 
```

kakopappa, 2017-03-04:
Same on Windows

runaway76, 2017-03-04:
Can confirm the same Problem under Windows 10 64-bit.

Ligament, 2017-03-04:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.14.4 x64
**Electron**: 1.3.13
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 1.7.2

### Stack Trace

Failed to load the platformio-ide package

```
At Refused to evaluate a string as JavaScript because 'unsafe-eval' is not an allowed source of script in the following Content Security Policy directive: "script-src 'self'".

EvalError: Refused to evaluate a string as JavaScript because 'unsafe-eval' is not an allowed source of script in the following Content Security Policy directive: "script-src 'self'".

    at Ajv.localCompile (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:139:26)
    at Ajv.resolve (/packages/platformio-ide/node_modules/ajv/lib/compile/resolve.js:53:19)
    at Object.resolveRef (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:210:21)
    at /packages/platformio-ide/node_modules/ajv/lib/dotjs/ref.js:22:22)
    at /packages/platformio-ide/node_modules/ajv/lib/dotjs/validate.js:156:37)
    at /packages/platformio-ide/node_modules/ajv/lib/dotjs/properties.js:199:26)
    at generate_validate (/packages/platformio-ide/node_modules/ajv/lib/dotjs/validate.js:229:37)
    at localCompile (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:99:22)
    at Ajv.compile (/packages/platformio-ide/node_modules/ajv/lib/compile/index.js:67:13)
    at _compile (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:328:29)
    at getSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:192:51)
    at validate (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:91:11)
    at validateSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:162:19)
    at _addSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:288:7)
    at addSchema (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:133:26)
    at addInitialSchemas (/packages/platformio-ide/node_modules/ajv/lib/ajv.js:396:39)
    at /packages/platformio-ide/node_modules/ajv/lib/ajv.js:78:3)
    at /packages/platformio-ide/node_modules/har-validator/lib/node4/promise.js:43:11)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/platformio-ide/node_modules/request/lib/har.js:5:16)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/platformio-ide/node_modules/request/request.js:25:31)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/platformio-ide/node_modules/request/index.js:144:19)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/platformio-ide/lib/versions/view.js:22:42)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
[... 26 lines cut ...]
    at PackageManager.module.exports.PackageManager.loadPackage (/app.asar/src/package-manager.js:468:14)
    at PackageManager.module.exports.PackageManager.activatePackage (/app.asar/src/package-manager.js:548:30)
    at /app.asar/node_modules/settings-view/lib/package-manager.js:439:29
    at exit (/app.asar/node_modules/settings-view/lib/package-manager.js:69:16)
    at triggerExitCallback (/app.asar/src/buffered-process.js:303:11)
    at /app.asar/src/buffered-process.js:333:11)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

### Commands

```
  2x -6:52.4.0 platformio-ide:initialize-new-project (atom-pane.pane.active)
     -6:44 core:paste (input.hidden-input)
```

### Non-Core Packages

```
platformio-ide 1.7.2 
```

me use Windows 10 64bit

mmaxus35, 2017-03-04:
Same issue on Windows 10 64bit

ivankravets, 2017-03-04:
The issue is linked with broken the latest version of `request` package that has been released 7 hours ago.

We will release v1.7.3 PIO IDE and fix it to the previous version of `request`.

phito, 2017-03-04:
Anyone found a workaround?

ivankravets, 2017-03-04:
Please wait for a few minutes... I'm working on the hot fix...

phito, 2017-03-04:
Oh alright, I didn't understand that you were doing a hotfix! Thanks for the fast reaction

ivankravets, 2017-03-04:
Done! Please open **Atom Package Manager**:

* Windows, Menu: File > Settings
* macOS, Menu: Atom > Preferences
* Linux, Menu: Edit > Preferences

1. Settings > Packages > Uninstall current broken 1.7.2 
2. Settings > Install > Search for `platformio-ide` and install 1.7.3.

P.S: Don't forget to try PlatformIO IDE 2.0 😊

closed 2017-03-04 by ivankravets

RuBiCK, 2017-03-05:
I had the same error but now it works with new version 1.7.3

commit 1fd05937 (2017-03-04): Notify about outdated Atom; Revert to the previous version of `request` package and resolved #249