platformio/platformio-atom-ide #251: InstallationManager

Rodrigoreis51, 2017-03-04:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\Rodrigo\AppData\Local\Temp\tmp-7172abVOh6S6hBEu\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Rodrigo\AppData\Local\Temp\tmp-7172abVOh6S6hBEu\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Rodrigo\AppData\Local\Temp\tmp-7172abVOh6S6hBEu\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Rodrigo\AppData\Local\Temp\tmp-7172abVOh6S6hBEu\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\Rodrigo\AppData\Local\Temp\tmp-7172abVOh6S6hBEu\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\Rodrigo\.at...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-03-04:
Do you see this error when restart Atom?

ivankravets, 2017-03-09:
Please upgrade to the latest version

closed 2017-03-09 by ivankravets