platformio/platformio-atom-ide #252: InstallationManager

alexandreleick, 2017-03-06:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/wy/3ttxv4_j75nbg6dtpkt_y0qc0000gn/T/tmp-4459FB64mYHO9f7H/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/wy/3ttxv4_j75nbg6dtpkt_y0qc0000gn/T/tmp-4459FB64mYHO9f7H/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/wy/3ttxv4_j75nbg6dtpkt_y0qc0000gn/T/tmp-4459FB64mYHO9f7H/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/alexandre/.atom/packages/platformio-ide/penv/lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/Users/alexandre/.atom/packages/platformio-ide/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-03-06:
What do you see after Atom restart?

alexandreleick, 2017-03-06:
It's good !
Thanks you

closed 2017-03-09 by ivankravets