platformio/platformio-atom-ide #253: InstallationManager

icaven, 2017-03-06:
PIP: /Users/xx/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#snimissingwarning.
  SNIMissingWarning
/Users/xx/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

/Users/xx/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

/Users/xx/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

Command "/Users/xx/.atom/packages/platformio-ide/penv/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/cm/38_m31112qqd32ngxjj_vkbm0000gn/T/pip-build-zSNaQe/bottle/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/cm/38_m31112qqd32ngxjj_vkbm0000gn/T/pip-9tGFBh-record/install-record.txt --single-version-externally-managed --compile --install-headers /Users/xx/.atom/packages/platformio-ide/penv/bin/../include/site/python2.7/bottle" failed with error code 1 in /private/var/folders/cm/38_m31112qqd32ngxjj_vkbm0000gn/T/pip-build-zSNaQe/bottle/

This occurred when installing the PlatformIO Preview 2.0 on Atom 1.14 and Mac OS X 10.8.5

ivankravets, 2017-03-10:
1. Atom: Settings ⚙️ > Packages
2. Uninstall `platformio-ide` packages
3. Atom: Settings ⚙️ > Install > `platformio-ide`
4. IDE > Home > Switch to PlatformIO IDE 2.0
5. Does it work now?

icaven, 2017-03-13:
I followed the steps above, and no errors were reported this time.

closed 2017-03-13 by icaven

ivankravets, 2017-03-13:
Thanks a lot for the report!