platformio/platformio-atom-ide #254: InstallationManager

WillGreen98, 2017-03-06:
Error: Error: ENOENT: no such file or directory, open '/Users/Will/.atom/packages/platformio-ide/.cache/examples.tar.gz'

ivankravets, 2017-03-09:
Could you reproduce this issue after Atom restart?

WillGreen98, 2017-03-09:
Yes, why? I have fixed the issue now though.

ivankravets, 2017-03-09:
How did you resolve it?

WillGreen98, 2017-03-10:
What OS are you using?

ivankravets, 2017-03-10:
I need information to fix this issue. What is your OS? Do you use Proxy?

WillGreen98, 2017-03-10:
I use macOS and no, you do not need a proxy for platformio.

Will.

ivankravets, 2017-03-10:
And how did you fix it?

WillGreen98, 2017-03-10:
For macOS, I used homebrew to install the dependencies needed for platformio to be executed, I then linked those files to the directory of platformio, I installed core, terminal and IDE within atom and then restarted. Worked for me.

Will.

ivankravets, 2017-03-10:
Wow! You don't need to do that. PlatformIO IDE has built-in PIO Core.

1. Try uninstall HomeBrew's PIO Core (`brew uninstall platformio`)
2. Remove all tracks of PIO Core in a system
3. Atom: Settings ⚙️ > Packages
4. Uninstall `platformio-ide` packages
5. Atom: Settings ⚙️ > Install > `platformio-ide`
6. IDE > Home > Switch to PlatformIO IDE 2.0
7. Does it work now?

It should work now. Please check that "Built-in PIO Core" is enabled in PIO IDE settings. Menu> PlatformIO > Settings > PlatformIO IDE

WillGreen98, 2017-03-10:
I will try that for compiler errors but what I did worked for linking libraries and took like 30 seconds. For running and uploading code I will try your method. Thank you.

Will.

WillGreen98, 2017-03-10:
I cannot see Built-in PIO Core in the settings.

Will.

From: Ivan Kravets <notifications@github.com>
Reply-To: platformio/platformio-atom-ide <reply@reply.github.com>
Date: Friday, 10 March 2017 at 11:28
To: platformio/platformio-atom-ide <platformio-atom-ide@noreply.github.com>
Cc: Will Green <will.green98@hotmail.com>, Author <author@noreply.github.com>
Subject: Re: [platformio/platformio-atom-ide] InstallationManager (#254)

Built-in PIO Core

ivankravets, 2017-03-10:
Settings ⚙️ Packages > PlatformIO IDE > "use built-in PIO Core" shoud be checked

WillGreen98, 2017-03-10:
Yes, that is checked, I have just found it.

Will.

From: Ivan Kravets <notifications@github.com>
Reply-To: platformio/platformio-atom-ide <reply@reply.github.com>
Date: Friday, 10 March 2017 at 11:28
To: platformio/platformio-atom-ide <platformio-atom-ide@noreply.github.com>
Cc: Will Green <will.green98@hotmail.com>, Author <author@noreply.github.com>
Subject: Re: [platformio/platformio-atom-ide] InstallationManager (#254)

Built-in PIO Core

ivankravets, 2017-03-10:
Now please perform all steps from https://github.com/platformio/platformio-atom-ide/issues/254#issuecomment-285646404

I've deployed new PIO IDE 2.0 alpha build yesterday.

WillGreen98, 2017-03-10:
Still getting upload errors, I wanted to use platformio for the autocorrect and intellisense but I might just use the arduino IDE.
Want this to work…Urg.

Will.

ivankravets, 2017-03-17:
https://community.platformio.org/t/urgent-platformio-ide-2-0-preview-doesnt-receive-any-updates/1620/11

closed 2017-03-17 by ivankravets