platformio/platformio-atom-ide #257: InstallationManager

fuinha, 2017-03-09:
Virtualenv Download: Traceback (most recent call last):
  File "/tmp/tmp-5221T2HAuIlDyfri/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/tmp/tmp-5221T2HAuIlDyfri/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/tmp/tmp-5221T2HAuIlDyfri/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/fuinhavaio/.atom/packages/platformio-ide/penv/lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/home/fuinhavaio/.atom/packages/platformio-ide/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-03-09:
Could you reproduce this issue after Atom restart?

ivankravets, 2017-03-10:
1. Atom: Settings ⚙️ > Packages
2. Uninstall `platformio-ide` packages
3. Atom: Settings ⚙️ > Install > `platformio-ide`
4. IDE > Home > Switch to PlatformIO IDE 2.0
5. Does it work now?

ivankravets, 2017-03-10:
Please reopen if you still have problems

closed 2017-03-10 by ivankravets

fuinha, 2017-03-10:
>
>    1. Atom: Settings ⚙️ > Packages
>    2. Uninstall platformio-ide packages
>    3. Atom: Settings ⚙️ > Install > platformio-ide
>    4. IDE > Home > Switch to PlatformIO IDE 2.0
>    5. Does it work now?
>
> I am testing
thanks

fuinha, 2017-03-10:
Ok thank you very much
-- 
*          Templus Eletronica LTDA*
*                    Marcelo  Anjos*

www.projetoslinux.com         +55 11 9473-82221
 projetoslinux@gmail.com      +55 11 9661-50250
  11101485@unifieo.edu.br   +1 347 279-0632
<projetoslinux@gmail.com>

fuinha, 2018-12-04:
Olá,

Eu acabei de assinar o abaixo-assinado "Justiça! Punição para quem matou o
cachorro do CARREFOUR de Osasco" e queria saber se você pode ajudar
assinando também.

A nossa meta é conseguir 260.326 assinaturas e precisamos de mais apoio.
Você pode ler mais sobre este assunto e assinar o abaixo-assinado aqui:

http://chng.it/ZNqkpfnqZV

Obrigado!
Marcelo anjos