platformio/platformio-atom-ide #259: InstallationManager

ChCh27, 2017-03-10:
PIP: 
Usage:   
  pip install [options] <requirement specifier> ...
  pip install [options] -r <requirements file> ...
  pip install [options] [-e] <vcs project url> ...
  pip install [options] [-e] <local project path> ...
  pip install [options] <archive url/path> ...

no such option: --no-cache-dir

ivankravets, 2017-03-10:
1. What is your OS?
2. Please provide an output of `pip --version` in system terminal

ChCh27, 2017-03-10:
Centos 7.2
$ pip -V
pip 9.0.1 from /usr/lib/python2.7/site-packages (python 2.7)
but!
in penv:
$ bin/pip -V
pip 1.4.1 from /home/chch/.platformio/penv/lib/python2.7/site-packages (python 2.7)

ivankravets, 2017-03-10:
Do you have global `virtualenv --version` in a system? What is a version?

ivankravets, 2017-03-10:
Please re-test:

1. Atom: Settings ⚙️ > Packages
2. Uninstall `platformio-ide` packages
3. Atom: Settings ⚙️ > Install > `platformio-ide`
4. IDE > Home > Switch to PlatformIO IDE 2.0
5. Does it work now?

closed 2017-03-10 by ivankravets

ChCh27, 2017-03-10:
I see only "Try PlatformIO IDE 2.0 Preview" .

ChCh27, 2017-03-10:
I've "tried" .
It works now. 
Thank you!

commit 91758307 (2017-03-10): Don’t use ‘--no-cache-dir’ option for old version of PIP // Resolve #259