platformio/platformio-atom-ide #260: Linter Repeatedly Reindexes

mrkaroshi, 2017-03-14:
I've got an issue whereby the Linter 'reindexing' message is coming up over and over again, and now I'm getting 'no such file or directory' errors for frameworks I know are installed.

One clue might be that when I open the Developer Console I get this message:

> Failed to clear temp storage: It was determined that certain files are unsafe for access within a Web application, or that too many calls are being made on file resources. SecurityError

I believe the error coincides with when I created a new .cpp file in my library folder called 'draft.cpp' that I was just using to try out a redraft of my contributions. Any clues as to what this might be?

mrkaroshi, 2017-03-14:
It also seems to be tied into the Python issue...
[image]

mrkaroshi, 2017-03-14:
The issue may be to do with dependencies also, as Linter appears to be repeatedly indexing the same things. Removing 'lib/Adafruit_BNO055' from my platformio.ini stopped the repeated indexing but now Build appears to be building twice and trying to upload twice.
[image]

ivankravets, 2017-03-14:
1. OS
2. PIO IDE Version
3. PIO Core version
4. Atom version

mrkaroshi, 2017-03-14:
Sorry, knew I'd forgotten something, put the info in the revised issue.

ivankravets, 2017-03-15:
> IDE 1.7.3,

Please switch to PlatformIO IDE 2.0.
Menu: PlatformIO > Home > Switch to PIO IDE 2.0

mrkaroshi, 2017-03-15:
Seems to be working well, I'll post at the end of the day - if nothing happens I think we can close. Thank you for your support!

mrkaroshi, 2017-03-17:
Problem solved, think this can be closed, many thanks!

closed 2017-03-25 by ivankravets