platformio/platformio-atom-ide #267: Stuck at "PlatformIO IDE: Installing ..." after install via apm & keeps opening processes

dmte, 2017-03-17:
I'm using Windows 10 and after installing PlatformIO IDE via apm (I also tried installation from within Atom) the atom startup gets stuck at this loading screen:

[image]

When checking the taskmanager you can see hundereds of python processes:

[image]
[image]

ivankravets, 2017-03-25:
Please try `apm install --production platformio/platformio-atom-ide`

ivankravets, 2017-05-06:
Should be fixed in the latest PIO IDE 2.0 Beta 3

closed 2017-05-06 by ivankravets