platformio/platformio-atom-ide #270: InstallationManager

2610, 2017-03-18:
PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x107b45c10>: Failed to establish a new connection: [Errno 8] nodename nor servname provided, or not known',)': /platformio/platformio/archive/develop.zip

  Retrying (Retry(total=3, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x107b45d50>: Failed to establish a new connection: [Errno 8] nodename nor servname provided, or not known',)': /platformio/platformio/archive/develop.zip

  Retrying (Retry(total=2, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x107b45e90>: Failed to establish a new connection: [Errno 8] nodename nor servname provided, or not known',)': /platformio/platformio/archive/develop.zip

  Retrying (Retry(total=1, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x107b45fd0>: Failed to establish a new connection: [Errno 8] nodename nor servname provided, or not known',)': /platformio/platformio/archive/develop.zip

  Retrying (Retry(total=0, connect=None, read=None, redirect=None)) after connection broken by 'NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x107c1f150>: Failed to establish a new connection: [Errno 8] nodename nor servname provided, or not known',)': /platformio/platformio/archive/develop.zip

Exception:
Traceback (most recent call last):
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/Users/home/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 487, in send
    raise ConnectionError(e, request=request)
ConnectionError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /platformio/platformio/archive/develop.zip (Caused by NewConnectionError('<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x107c1f250>: Failed to establish a new connection: [Errno 61] Connection refused',))

ivankravets, 2017-03-23:
> VerifiedHTTPSConnection ::  Failed to establish a new connection

What is your OS?

ivankravets, 2017-03-25:
You have problem with HTTPS/Internet connection

closed 2017-03-25 by ivankravets