platformio/platformio-atom-ide #277: InstallationManager

xiaalan123, 2017-03-22:
Error: Error: 
gyp info it worked if it ends with ok
gyp info using node-gyp@3.4.0
gyp info using node@4.4.5 | win32 | x64
gyp http GET https://atom.io/download/electron/v1.3.13/iojs-v1.3.13.tar.gz
gyp WARN install got an error, rolling back install
gyp ERR! install error 
gyp ERR! stack Error: connect ETIMEDOUT 54.231.120.107:443
gyp ERR! stack     at Object.exports._errnoException (util.js:870:11)
gyp ERR! stack     at exports._exceptionWithHostPort (util.js:893:20)
gyp ERR! stack     at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1061:14)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Users\\Administrator\\AppData\\Local\\atom\\app-1.15.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Administrator\\AppData\\Local\\atom\\app-1.15.0\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "install" "--runtime=electron" "--target=1.3.13" "--dist-url=https://atom.io/download/electron" "--arch=x64" "--ensure"
gyp ERR! cwd C:\Users\Administrator\.atom
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok

ivankravets, 2017-03-23:
Is something wrong with the Internet connection?

closed 2017-03-25 by ivankravets

jmszwzr, 2017-06-01:
Allow the firewall through port 443