platformio/platformio-atom-ide #279: Uncaught Error: ENOENT: no such file or directory, open '$ATOM_HOME/packages\platformio-ide\lib\i...

sim2egor, 2017-03-23:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.15.0 x64
**Electron**: 1.3.13
**OS**: Unknown Windows version
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 1.7.3

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, open 'C:\Users\Igor\.atom\packages\platformio-ide\lib\init\template.html'

```
At fs.js:640

Error: ENOENT: no such file or directory, open 'C:\Users\Igor\.atom\packages\platformio-ide\lib\init\template.html'
    at Error (native)
    at Object.fs.openSync (fs.js:640:18)
    at Object.module.(anonymous function) [as openSync] (ELECTRON_ASAR.js:168:20)
    at Object.fs.readFileSync (fs.js:508:33)
    at Object.fs.readFileSync (ELECTRON_ASAR.js:501:29)
    at InitializeNewProjectView.buildElement (/packages/platformio-ide/lib/base-view.js:51:44)
    at InitializeNewProjectView.BaseView (/packages/platformio-ide/lib/base-view.js:37:25)
    at /packages/platformio-ide/lib/init/view.js:47:91)
    at Object.command (/packages/platformio-ide/lib/init/command.js:195:14)
    at /packages/platformio-ide/lib/main.js:67:59)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/app.asar/src/atom-environment.js:1162:28)
    at EventEmitter.outerCallback (/app.asar/src/application-delegate.js:312:25)
    at emitThree (events.js:116:13)
    at EventEmitter.emit (events.js:194:7)
```

### Commands

```
  2x -3:09.7.0 platformio-ide:project-examples (ol.tree-view.full-menu.list-tree.has-collapsable-children.focusable-panel)
     -0:37.7.0 application:add-project-folder (ol.tree-view.full-menu.list-tree.has-collapsable-children.focusable-panel)
     -0:21.2.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:15.5.0 platformio-ide:initialize-new-project (ol.tree-view.full-menu.list-tree.has-collapsable-children.focusable-panel)
```

### Non-Core Packages

```
atom-clock 0.1.6 
atom-ctags 5.0.0 
autocomplete-ctags 0.4.1 
build 0.67.0 
busy 0.7.0 
ctags-status 1.3.7 
platformio-ide 1.7.3 
platformio-ide-terminal 2.5.0 
tool-bar 1.1.0 
```

ivankravets, 2017-03-23:
Could you uninstall/install again?

DavidPowell, 2017-03-24:
I had the same issue, and eventually managed to resolve it by uninstalling and reinstalling platformio-ide.

FWIW, this happened when re-installing 1.7.3, to fix the problem with the 2.0.0 alphas which get stuck in an infinite reboot cycle.

Also potentially relevant is that my Python 2.7 is not installed system wide. I start atom from a specific windows terminal which has the right python environment loaded into the path. When platformio-ide is installed, it offers to restart atom. Unfortunately, when restarting itself, atom seems to lose all the environment information about environment variables.  The solution was to close atom, manually restart it from the command-line, and then it is possible to install platformio-ide with no errors.

ivankravets, 2017-03-24:
The latest Alpha of PIO IDE 2.0 doesn't require Python to be installed in a system. We haven't announced it yet because users can't install 2.0 directly without 1.0 (which requires Python).

I hope that the first public beta of PIO IDE 2.0 will be released next week. We've had huge refactoring this week. Whole IDE was entirely rewritten in React. See https://github.com/platformio/platformio-atom-ide/commit/fc6ddcd63922f4bce7a18d4d207db945b98eada1

@DavidPowell you try this:

1. Uninstall special Python that you install for PlatformIO
2. Uninstall `platformio-ide` package in Atom: Settings > Packages > search for `platformio-ide`.
3. Close Atom
4. Open `cmd` and run `apm install --production platformio/platformio-atom-ide`

New PIO IDE 2.0 will install custom Python 2.7 into isolated folder `~/.platformio/python27` and use it. It will not visible in your global scope and will not conflict with other programs.

DavidPowell, 2017-03-27:
@ivankravets I followed your instructions and everything works fine. Eliminating the extra step of installing python will be great for new users. Thanks for the great work.

ivankravets, 2017-05-06:
Should be fixed in the latest PIO IDE 2.0 Beta 3

closed 2017-05-06 by ivankravets