platformio/platformio-atom-ide #284: InstallationManager

lizardkings, 2017-03-27:
Error: Error: unexpected end of file
    at ProjectExamplesStage.<anonymous> (file:///C:/Users/dgt/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:148:13)
    at throw (<anonymous>)
    at step (C:\Users\dgt\.atom\packages\platformio-ide\lib\installer\stages\project-examples.js:13:273)

ivankravets, 2017-03-27:
Could you restart Atom? Does it work now?

lizardkings, 2017-03-27:
Restarted and working perfect now.

ivankravets, 2017-03-27:
Have you had PIO IDE installed before?

lizardkings, 2017-03-27:
No never on this machine. I had internet connection  issues during installation and first time startup.

ivankravets, 2017-03-27:
TODO: Need to check file size/checksum after downloading.

lizardkings, 2017-03-27:
No the download was ok! But after the download I lost my connection. I
installed without internet. Do not not if this information was important to
solve the issue.

ivankravets, 2017-03-27:
I got you. I mean that after PIO IDE downloading, PIO continues installation. In this case need to check Internet connection.

ivankravets, 2017-05-06:
@orgkhnargh Plese improve this function https://github.com/platformio/platformio-atom-ide/blob/develop/lib/installer/helpers.js#L28

1. Fetch `Content-Length` for each DL request.
2. After downloading check size with original
3. If it differs, try re-download file.
4. Add new extra option to this function called `retries=3`.

PR #451 (2017-05-08): Retry download when error occurs or when file size is incorrect
Resolve #284
commit 19ea3e8a (2017-05-08): Retry download when error occurs or when file size is incorrect (#451)

Resolve #284
closed 2017-05-08 by ivankravets