platformio/platformio-atom-ide #285: InstallationManager

patriciooliveira, 2017-03-27:
Error: Error: events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: ENOENT: no such file or directory, open 'C:\Users\EXPGAME\.atom\packages\linter-ui-default\node_modules\sb-config-file\lib\helpers.js'
    at Error (native)

    at AtomDependenciesStage.<anonymous> (file:///C:/Users/EXPGAME/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:164:13)
    at throw (<anonymous>)
    at step (C:\Users\EXPGAME\.atom\packages\platformio-ide\lib\installer\stages\atom-dependencies.js:15:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)

ivankravets, 2017-03-27:
Could you restart Atom? Does it work now?

ivankravets, 2017-04-30:
Depends on https://github.com/platformio/platformio-atom-ide/issues/424

commit 9bb32114 (2017-05-01): Install dependencies one by one when failed to install all at once

Resolve #424, Resolve #285, Resolve #288, Resolve #292, Resolve #310,
Resolve #314, Resolve #315, Resolve #331, Resolve #334, Resolve #338,
Resolve #345, Resolve #358, Resolve #362, Resolve #380, Resolve #383,
Resolve #387, Resolve #394, Resolve #397, Resolve #401, Resolve #418,
Resolve #421, Resolve #332, Resolve #339, Resolve #342, Resolve #346,
Resolve #363, Resolve #381, Resolve #386, Resolve #395, Resolve #402
closed 2017-05-01 by ivankravets