platformio/platformio-atom-ide #287: InstallationManager

dcirne, 2017-03-28:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/9q/n0h9l8cd1rj87stpy3zd0x6w0000gn/T/tmp-2637ffWAE2IoRrv5/virtualenv-15.1.0/virtualenv.py", line 2328, in <module>
    main()
  File "/var/folders/9q/n0h9l8cd1rj87stpy3zd0x6w0000gn/T/tmp-2637ffWAE2IoRrv5/virtualenv-15.1.0/virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "/var/folders/9q/n0h9l8cd1rj87stpy3zd0x6w0000gn/T/tmp-2637ffWAE2IoRrv5/virtualenv-15.1.0/virtualenv.py", line 925, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/dalmo/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 360, in abspath
    if not isabs(path):
  File "/Users/dalmo/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-03-28:
Could you remove `/Users/dalmo/.platformio/penv` folder and restart IDE?

dcirne, 2017-03-28:
Hi @ivankravets I tried what you suggested, but when the IDE was launched again, the same error appeared. 

Having said that, it seems that all works normally after you close the error message and continue using PlatformIO. For what I can tell the installation error is not preventing things from working as expected.

ivankravets, 2017-03-30:
How did you resolve that?

dcirne, 2017-03-30:
I didn't really resolve it. Since it kept showing the error every time I launched the IDE, I decided to remove the installation and re-install a fresh copy. Since then it has been working with no errors. I closed the issue because I didn't want to keep clutter among the open issues.

ivankravets, 2017-03-30:
We should fix this issue, that is not normal behaviour.

1. Please navigate to `/Users/dalmo/.atom/packages/platformio-ide/lib/installer/stages/platformio-core.js`, open with any text editor
2. Replace [this line](https://github.com/platformio/platformio-atom-ide/blob/develop/lib/installer/stages/platformio-core.js#L26) with 
```
static vitrualenvUrl = 'https://pypi.python.org/packages/source/v/virtualenv/virtualenv-14.0.6.tar.gz';
```
3. Remove `/Users/dalmo/.platformio/penv`
4. Start Atom

ivankravets, 2017-03-30:
Opsss, sorry, I missed " Since then it has been working with no errors"

closed 2017-03-30 by ivankravets