platformio/platformio-atom-ide #288: Install missed packages: platformio-ide-terminal,busy-signal,linter-ui-default,intentions

basbaba, 2017-03-28:
Error: events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: ENOENT: no such file or directory, open '/Users/victor/.atom/packages/platformio-ide-terminal/keymaps/platformio-ide-terminal.cson'
    at Error (native)

    at /Users/victor/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:86:27
    at onExit (/Users/victor/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/Users/victor/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/Users/victor/app/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:299:11)
    at /Users/victor/app/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:312:11
    at Socket.<anonymous> (/Users/victor/app/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:181:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

ivankravets, 2017-03-28:
1. Have you had PlatformIO IDE installed before?
2. Try to uninstall `platformio-ide-terminal` and install again

Settings :gear: > Pakcages > Search for platformio-ide-terminal

ivankravets, 2017-04-30:
Depends on https://github.com/platformio/platformio-atom-ide/issues/424

commit 9bb32114 (2017-05-01): Install dependencies one by one when failed to install all at once

Resolve #424, Resolve #285, Resolve #288, Resolve #292, Resolve #310,
Resolve #314, Resolve #315, Resolve #331, Resolve #334, Resolve #338,
Resolve #345, Resolve #358, Resolve #362, Resolve #380, Resolve #383,
Resolve #387, Resolve #394, Resolve #397, Resolve #401, Resolve #418,
Resolve #421, Resolve #332, Resolve #339, Resolve #342, Resolve #346,
Resolve #363, Resolve #381, Resolve #386, Resolve #395, Resolve #402
closed 2017-05-01 by ivankravets