platformio/platformio-atom-ide #291: Platform command failed: platform install atmelsam

Faynman, 2017-03-28:
Error: Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x0000000003EB1F28>> ignored
Exception AttributeError: "'FileDownloader' object has no attribute '_request'" in <bound method FileDownloader.__del__ of <platformio.downloader.FileDownloader object at 0x0000000003F35A90>> ignored

Error: Could not install 'atmelsam' with version requirements '*' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

    at utils.runPIOCommand (helpers.js:61:25)
    at onExit (utils.js:92:7)
    at exit (utils.js:102:21)
    at triggerExitCallback (C:\Users\Faynman\AppData\Local\atom\app-1.16.0-beta0\resources\app.asar\src\buffered-process.js:265:9)
    at ChildProcess.process.on.code (C:\Users\Faynman\AppData\Local\atom\app-1.16.0-beta0\resources\app.asar\src\buffered-process.js:295:9)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

ivankravets, 2017-03-28:
Internet connection? Could you try again?

ivankravets, 2017-04-30:
Please reopen if you still have problem

closed 2017-04-30 by ivankravets