platformio/platformio-atom-ide #292: Install missed packages: busy-signal,linter-ui-default,intentions

EranGoldman, 2017-03-28:
Error: events.js:141

      throw er; // Unhandled 'error' event

      ^

Error: ENOENT: no such file or directory, open '/home/marinetech/.atom/packages/linter-ui-default/node_modules/core-js/fn/_.js'

    at Error (native)

    at /home/marinetech/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:86:27
    at onExit (/home/marinetech/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/home/marinetech/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/opt/atom/resources/app.asar/src/buffered-process.js:322:11)
    at ChildProcess.<anonymous> (/opt/atom/resources/app.asar/src/buffered-process.js:352:11)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

ivankravets, 2017-03-28:
Could you try restart Atom?

emranweb, 2017-04-29:
I have some problem in writing JS code. After that I disable in Plugin. Now It Working Fine.

ivankravets, 2017-04-30:
Depends on https://github.com/platformio/platformio-atom-ide/issues/424

ivankravets, 2017-04-30:
@emranweb What did you disable?

commit 9bb32114 (2017-05-01): Install dependencies one by one when failed to install all at once

Resolve #424, Resolve #285, Resolve #288, Resolve #292, Resolve #310,
Resolve #314, Resolve #315, Resolve #331, Resolve #334, Resolve #338,
Resolve #345, Resolve #358, Resolve #362, Resolve #380, Resolve #383,
Resolve #387, Resolve #394, Resolve #397, Resolve #401, Resolve #418,
Resolve #421, Resolve #332, Resolve #339, Resolve #342, Resolve #346,
Resolve #363, Resolve #381, Resolve #386, Resolve #395, Resolve #402
closed 2017-05-01 by ivankravets