platformio/platformio-atom-ide #294: Problem has beenn occured while uprading PlatformIO IDE.

limawill, 2017-03-29:
Error: undefined

    at utils.runAPMCommand.busyTitle (/home/will/.atom/packages/platformio-ide/lib/installer/ide.js:111:13)
    at onExit (/home/will/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/home/will/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/usr/lib64/atom/src/buffered-process.js:303:11)
    at /usr/lib64/atom/src/buffered-process.js:324:11
    at Socket.<anonymous> (/usr/lib64/atom/src/buffered-process.js:185:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

ivankravets, 2017-03-29:
Please restart Atom. Which version of PIO IDE do you see on PIo Home screen?

ivankravets, 2017-04-30:
@orgkhnargh if an error occurs while automatically upgrading PIO IDE, need to ask the user upgrade IDE manually (modal window) \.

commit bebf207b (2017-05-04): Print detailed information about PIO IDE Upgrade when error has occurred // Resolve #294
closed 2017-05-04 by ivankravets