platformio/platformio-atom-ide #295: InstallationManager

mainephd, 2017-03-29:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/yv/0z1k9z255nl2plpk4wb8g_sw0000gq/T/tmp-87003oEDrL3vEl6Ii/virtualenv-15.1.0/virtualenv.py", line 2328, in <module>

    main()
  File "/var/folders/yv/0z1k9z255nl2plpk4wb8g_sw0000gq/T/tmp-87003oEDrL3vEl6Ii/virtualenv-15.1.0/virtualenv.py", line 713, in main

    symlink=options.symlink)
  File "/var/folders/yv/0z1k9z255nl2plpk4wb8g_sw0000gq/T/tmp-87003oEDrL3vEl6Ii/virtualenv-15.1.0/virtualenv.py", line 925, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/rkk0035/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/Users/rkk0035/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-03-29:
Do you see this error after Atom restart?

mainephd, 2017-04-01:
The error goes away after the restart of atom @ivankravets

closed 2017-04-30 by ivankravets

M0derator, 2017-06-14:
Did anyone get this fixed? I am having the same error after Sophos AV decided to delete the requirements.txt file which I could not reinstall for some reason.

ivankravets, 2017-06-14:
@M0derator try 
```
sudo pip install -U pip setuptools
rm -rf ~/.platformio/penv
```

Start Atom.

M0derator, 2017-06-14:
The directory '/home/itwasluckvpn/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/itwasluckvpn/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.

This is the error I'm getting, I kind of know its related to Sophos AV to removing the file. I tried removing Sophos AV and giving it another go but using the commands you provide I'm getting this. Thanks for the quick reply BTW, much appreciated!

commit 61d7e67b (2017-04-30): Revert back virtualenv to 14.0.6 // Resolve #276, Resolve #286, Resolve #295, Resolve #340, Resolve #382, Resolve #388, Resolve #391, Resolve #405, Resolve #407, Resolve #411, Resolve #420