platformio/platformio-atom-ide #296: Problem has beenn occured while uprading PlatformIO IDE.

bluefix, 2017-03-29:
Error: events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: ENOENT: no such file or directory, open '/Users/blue/.atom/packages/platformio-ide/node_modules/tmp/lib/tmp.js'
    at Error (native)

    at utils.runAPMCommand.busyTitle (/Users/blue/.atom/packages/platformio-ide/lib/installer/ide.js:154:87)
    at onExit (/Users/blue/.atom/packages/platformio-ide/lib/utils.js:144:7)
    at exit (/Users/blue/.atom/packages/platformio-ide/lib/utils.js:159:14)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:322:11)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:343:11
    at Socket.<anonymous> (/Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:204:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

ivankravets, 2017-03-30:
Which version of PIO IDE do you use now?

ivankravets, 2017-04-30:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/294

closed 2017-04-30 by ivankravets