platformio/platformio-atom-ide #298: InstallationManager fails behind proxy

josephwinston, 2017-03-29:
A clean install on Windows 7 with conda

```
Conda Virtualenv: An unexpected error has occurred.
Please consider posting the following information to the
conda GitHub issue tracker at:

    https://github.com/conda/conda/issues

Current conda install:

               platform : win-64
          conda version : 4.2.13
       conda is private : False
      conda-env version : 4.2.13
    conda-build version : not installed
         python version : 2.7.13.final.0
       requests version : 2.12.4
       root environment : C:\Users\hb55683\AppData\Local\Continuum\Anaconda2  (writable)
    default environment : C:\Users\hb55683\AppData\Local\Continuum\Anaconda2
       envs directories : C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\envs
          package cache : C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\pkgs
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.continuum.io/pkgs/free/win-64
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/pro/win-64
                          https://repo.continuum.io/pkgs/pro/noarch
                          https://repo.continuum.io/pkgs/msys2/win-64
                          https://repo.continuum.io/pkgs/msys2/noarch
            config file : d:\home\josephwinston\.condarc
           offline mode : False

`$ C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\Scripts\conda-script.py create --yes --quiet python=2 --prefix C:\Users\hb55683\.platformio\penv`

    Traceback (most recent call last):
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\exceptions.py", line 479, in conda_exception_handler
        return_value = func(*args, **kwargs)
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\cli\main.py", line 145, in _main
        exit_code = args.func(args, p)
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\cli\main_create.py", line 68, in execute
        install(args, parser, 'create')
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\cli\install.py", line 238, in install
        prefix=prefix)
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\api.py", line 24, in get_index
        index = fetch_index(channel_urls, use_cache=use_cache, unknown=unknown)
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\fetch.py", line 306, in fetch_index
        for url in urls]
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\fetch.py", line 75, in func
        res = f(*args, **kwargs)
      File "C:\Users\hb55683\AppData\Local\Continuum\Anaconda2\lib\site-packages\conda\fetch.py", line 141, in fetch_repodata
        raise CondaRuntimeError("Invalid index file: {0}: {1}".format(join_url(url, filename), e))
    CondaRuntimeError: Runtime error: Invalid index file: https://conda.anaconda.org/conda-forge/win-64/repodata.json: Failed to parse: proxy.mycomp.com:80;PROXY

Report a problem

```

josephwinston, 2017-03-29:
Adding to ```.condarc```:

```
proxy_servers:
    http: http://localhost:3128
    https: http://localhost:3128
```

produces the same traceback

This is what I normally use since I run cntlm.

closed 2017-06-28 by ivankravets

ivankravets, 2017-06-28:
1. Please open Atom > Settings > Packages > PlatformIO IDE
2. Check "Advanced > Use development version of PlatformIO IDE"
3. Wait while Atom will re-install PIO IDE

Does it work now?

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298