platformio/platformio-atom-ide #300: InstallationManager

arendst, 2017-03-30:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\Theo\AppData\Local\Temp\tmp-7072lFLoenLqTUNo\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "C:\Users\Theo\AppData\Local\Temp\tmp-7072lFLoenLqTUNo\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "C:\Users\Theo\AppData\Local\Temp\tmp-7072lFLoenLqTUNo\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "C:\Users\Theo\AppData\Local\Temp\tmp-7072lFLoenLqTUNo\virtualenv-15.1.0\virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\Theo\AppData\Local\Temp\tmp-7072lFLoenLqTUNo\virtualenv-15.1.0\virtualenv.py", line 753, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Python27\Lib\subprocess.py", line 710, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 958, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

arendst, 2017-03-30:
Solved after upgrading python pip (9.0.1) and virtualenv (15.1.0)

closed 2017-03-30 by arendst

ivankravets, 2017-03-30:
What was a previous version of PIP/Virtualenv?

arendst, 2017-03-30:
Ah I'm not sure but I believe PIP was somewhere around 8.0 and Virtualenv was somewhere around 1.x

Sorry.

ivankravets, 2017-03-30:
According to your log, you didn't have `virtualenv` before. if PIO IDE Installer sees `virtualenv` in a system it uses it. However, in your case PIO IDE tried to download the latest version automatically.

In any case, thanks a lot that mentioned your solution.