platformio/platformio-atom-ide #302: Terminal: Uncaught TypeError: Cannot read property 'element' of undefined

ivankravets, 2017-03-30:
See https://github.com/platformio/platformio-atom-ide-terminal/issues/37#issuecomment-290381077

manugildev, 2017-04-02:
Seems to be solved in the new update. I've been playing around for hours and I didn't get any duplicate terminal that can't be closed.

ivankravets, 2017-04-02:
Thanks, @manugildev!

/c @squix78 @mrkaroshi @kalsan @dansmith65 @bcanustundag @00SteinsGate00 @hwacookie @vinodsr @glumb @spicajames  

Could you confirm that?

kalsan, 2017-04-03:
Currently I have no Serial Monitor at all, so "no cannot reproduce" due to completely non-functional serial monitor. Will look for another issue or create one to address the issue.

closed 2017-06-01 by ivankravets

dansmith65, 2017-06-01:
Sorry I never got back to you; I haven't used Platformio in months. Thanks for fixing this! (I'm assuming it's fixed; still haven't tested)

andrew-davie, 2017-07-27:
I get a similar problem. I'm using PlatformIO with an Arduino Micro.  What's happening is that the Micro has a software serial port, so when you do an upload, the port is removed and then reattached.  The port number for the USB increases (i.e., goes from something like /dev/cu.usbmodem128 to /dev/cu.usbmodem129) but the serial/terminal window is still using the previous ID (ie: /dev/cu.usbmodem128) and fails with this error.  If you open a serial terminal and manually select /dev/cu.usbmodem129 then it works. It's a pain because it breaks the automatic build/upload process.  I hope this helps.

realjax, 2018-07-25:
Still have this problem. Several 'unclosable' terminals. When trying to close them nothing happens, but when clicking the 'normal' icon, the error below is thrown.
Running  core 3.6.0a10 on osx 10.13.6

`TypeError: Cannot read property 'element' of undefined
    at PlatformIOTerminalView.module.exports.PlatformIOTerminalView.adjustHeight (/Users/username/.atom/packages/platformio-ide-terminal/lib/view.coffee:426:16)
    at PlatformIOTerminalView.module.exports.PlatformIOTerminalView.maximize (/Users/username/.atom/packages/platformio-ide-terminal/lib/view.coffee:203:8)
    at HTMLButtonElement.<anonymous> (/Users/username/.atom/packages/autocomplete-clang/node_modules/space-pen/lib/space-pen.js:220:36)
    at HTMLButtonElement.dispatch (/Users/username/.atom/packages/autocomplete-clang/node_modules/jquery/dist/jquery.js:4435:9)
    at HTMLButtonElement.elemData.handle (/Users/username/.atom/packages/autocomplete-clang/node_modules/jquery/dist/jquery.js:4121:28)`