platformio/platformio-atom-ide #303: Library command failed: lib --global show 39 --json-output

sorinescu, 2017-03-30:
Error: Error: no such option: --json-output

    at utils.runPIOCommand.spawnOptions (/Users/sorin/.atom/packages/platformio-ide/lib/library/helpers.js:39:25)
    at onExit (/Users/sorin/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/Users/sorin/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:322:11)
    at ChildProcess.<anonymous> (/Applications/Atom.app/Contents/Resources/app.asar/src/buffered-process.js:352:11)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

ivankravets, 2017-03-30:
It looks that you have multiple PIO Cores in a system. Please remove duplicates and use built-in PIO Core in PIO IDE.

Open system terminal (not PIO IDE) and run
```bash
pip uninstall platformio
# or if you used brew
brew uninstall platformio
```

Then, check that the latest PIO IDE 2.0beta2 is installed and IDE MENU: PlatformIO > Settings > PIO IDE > Ensure that "Use built-in PIo Core" is checked.

sorinescu, 2017-03-30:
Indeed, I uninstalled my existing platformio and installed the development version and everything works. Thanks!

closed 2017-03-30 by sorinescu