platformio/platformio-atom-ide #307: xtensa-lx106-elf-gcc is not a known internal or external command ...

FernandoGarcia, 2017-03-31:
Hi!

I have installed the IDE yesterday and I'm trying compile "project examples" for ESP8266 but I have always this error:

```
Compiling .pioenvs\esp01\src\blinky.o
'xtensa-lx106-elf-gcc' não é reconhecido como um comando interno
ou externo, um programa operável ou um arquivo em lotes.
*** [.pioenvs\esp01\src\blinky.o] Error 1
 [ERROR] Took 1.63 seconds
```

This result is for ESP8266-native sketch.

All packets are up to date.

```
PS C:\Users\Fernando Garcia\.platformio\project-examples\espressif\esp8266-native> platformio update
Updating tool-scons                      @ 3.20501.2      [Up-to-date]
 
Platform Manager
================
Platform Espressif 32
--------
Updating espressif32                     @ 0.6.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20000.0      [Up-to-date]

Updating toolchain-xtensa32              @ 1.50200.0      [Up-to-date]
Updating framework-arduinoespressif32    @ 1.1.2          [Up-to-date]
 
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.3.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Up-to-date]
Updating sdk-esp8266                     @ 1.10502.0      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
 
 
Library Manager
===============
PS C:\Users\Fernando Garcia\.platformio\project-examples\espressif\esp8266-native> platformio --version
PlatformIO, version 3.3.0
```

Can you help me please?

Best regards.

FernandoGarcia, 2017-03-31:
This error is for ESP8266-webserver sketch:

[image]

ivankravets, 2017-04-01:
It seems that problem is linked with a space in a user name. Please move this issue to appropriate repository https://github.com/platformio/platformio-core/issues

closed 2017-04-01 by ivankravets