platformio/platformio-atom-ide #308: 'Goto Declaration' doesn't work in simple project

Spongman, 2017-03-31:
[project.zip](https://github.com/platformio/platformio-atom-ide/files/886350/project.

- open 'project' folder in platformio ide
- open developer tools
- open Foo.cpp
- click of 'Foo' symbol, right-click 'Goto Declaration'

nothing happens.

developer tools console contains:
````
In file included from <stdin>:1:
C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:163:23: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint16_t PROGMEM port_to_mode_PGM[];
                      ^

autocomplete-clang.coffee:91 C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:113:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:98:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^

autocomplete-clang.coffee:91 In file included from <stdin>:1:
C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:164:23: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint16_t PROGMEM port_to_input_PGM[];
                      ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:113:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:98:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^

autocomplete-clang.coffee:91 In file included from <stdin>:1:
C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:165:23: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint16_t PROGMEM port_to_output_PGM[];
                      ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:113:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:98:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^
In file included from <stdin>:1:
C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:167:22: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint8_t PROGMEM digital_pin_to_port_PGM[];
                     ^

autocomplete-clang.coffee:91 C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:113:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:98:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^

autocomplete-clang.coffee:91 In file included from <stdin>:1:
C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:169:22: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint8_t PROGMEM digital_pin_to_bit_mask_PGM[];
                     ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:113:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:98:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^
In file included from <stdin>:1:
C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:170:22: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint8_t PROGMEM digital_pin_to_timer_PGM[];
                     ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:113:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:98:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^

autocomplete-clang.coffee:91 In file included from <stdin>:1:
In file included from C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:233:
In file included from C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino/USBAPI.h:27:
In file included from C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\util/delay.h:45:
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\util/delay_basic.h:108:5: error: invalid output constraint '=w' in asm
                : "=w" (__count)
                  ^

autocomplete-clang.coffee:91 In file included from <stdin>:1:
In file included from C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:233:
In file included from C:\Users\PiersH\.platformio\packages\framework-arduinoavr\cores\arduino/USBAPI.h:27:
C:\Users\PiersH\.platformio\packages\toolchain-atmelavr\avr\include\util/delay.h:112:3: warning: "Compiler optimizations disabled; functions from <util/delay.h> won't work as designed" [-W#warnings]
# warning "Compiler optimizations disabled; functions from <util/delay.h> won't work as designed"
  ^

autocomplete-clang.coffee:91 7 warnings and 1 error generated.
````

ivankravets, 2017-04-01:
Could you help us to debug this issue? We provide all information for [autocomplete-clang](https://github.com/yasuyuky/autocomplete-clang) package via `.clang_complete` file, it should be located in a project directory. That is known issue with Windows and embedded development platforms.

Thanks!

Spongman, 2017-04-01:
that file is included in the .zip i attached above.

Spongman, 2017-04-01:
maybe i have a newer clang than you?
````
C:\Users\User\.platformio\project>clang --version
clang version 4.0.0 (tags/RELEASE_400/final)
Target: x86_64-pc-windows-msvc
Thread model: posix
InstalledDir: C:\Program Files\LLVM\bin
````

Spongman, 2017-04-01:
this is definitely a clang issue. compiling the following file `clang -fsyntax-only foo.c`

````
void foo (short __count)
{
        __asm__ volatile (
                "absolute nonsense is ok here" "\n\t"
                : "but not here" (__count)
                : "0" (__count)
        );
}
````

results in
````
a.c:5:5: error: invalid output constraint 'but not here' in asm
                : "but not here" (__count)
                  ^
1 error generated.
````

so my version of clang is breaking on non-intel constraints.

how is this working on other OSes?

ivankravets, 2017-04-01:
NO no no no, that is a known issue. 
- https://community.platformio.org/t/platformio-ide-for-atom-go-to-declaration/98

Time, time, time... That is why I asked you to dig deeper and find a problem. Would be thankful.

ivankravets, 2017-04-01:
> how is this working on other OSes?

We have the same "errors" on Mac/LInux and it works! that is normal, Clang doesn't understand non-intel archs. Need to check `autocomplete-clang` package.

Spongman, 2017-04-01:
i'm not sure what more you want me to do. i found the problem. it doesn't work.

ivankravets, 2017-04-01:
I see that you have JS experience. Atom > Settings > Packages > autocomplete-clang > View code.

You can make some `console.log("my dump")` and check where it fails. Don't forget to reload Atom Window after each change. Atom doesn't reload scripts automatically. You can do that via `Menu: View > Developer > Reload Windows`.

Spongman, 2017-04-20:
this is fixed now.

closed 2017-04-20 by Spongman

ivankravets, 2017-04-21:
@Spongman thank you for PR! ;)