platformio/platformio-atom-ide #312: InstallationManager

ryanbbernstein, 2017-04-01:
PIP: 

That's all is says. Not able to use platformio at all.

ivankravets, 2017-04-01:
What is your OS?

ryanbbernstein, 2017-04-01:
Windows 10 64bit

None of the menu options seem to do anything. Doesn't give me any error. I have tried reinstalling through command line as well and it says failed.

ivankravets, 2017-04-01:
1. Do you have Python installed in a system? If yes, which version?
2. Did you use PIO IDE before?

ryanbbernstein, 2017-04-01:
Ya python 2.7
First time trying to use PIO

ivankravets, 2017-04-01:
Could you make clean uninstallation and install the latest Atom + stable PIO IDE 2.0.0-beta.2? Thanks!

* http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

ryanbbernstein, 2017-04-01:
Followed the instructions but still nothing.
Goes through installation with dependencies and core still flashing and then straight to this.
[image]

ivankravets, 2017-04-01:
Please mail me@ikravets.com, I'll explain further steps

ryanbbernstein, 2017-04-01:
Virtualenv=1.9.1

mapsi-fixi, 2017-04-26:
Hello, i have same problem. Windows 7 64bit. Python 2.7

ryanbbernstein, 2017-04-26:
Problem came from virtualenv being way out of date. Try updating to latest version and reinstalling PIO.

ivankravets, 2017-06-01:
Try in system terminal
```
pip install -U pip setuptools virtualenv
```

Restart Atom

closed 2017-06-01 by ivankravets