platformio/platformio-atom-ide #313: InstallationManager

mzalih, 2017-04-01:
Error: Could not get proxy from APM: npm config failed: 1

    at ProjectExamplesStage.<anonymous> (/Users/mzalih/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:148:13)
    at throw (<anonymous>)
    at step (/Users/mzalih/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:13:273)

ivankravets, 2017-04-01:
Do you see this error after Atom restart?

mzalih, 2017-04-13:
Yes I tried restarting the system, and reinstall atom too

ameisso, 2017-04-18:
Same problem here, 

I've tried to uncheck "Use Proxy Settings When Calling APM", but it doesn't change anything. 
Relaunching Atom doesn't eather.

ameanomes, 2017-05-02:
Didn't find a solution for this. Here is what i've tested : 
- removing atom and .atom folders 
- reinstall Atom 
- install older (1.14) version of atom 
- tried the Beta 
- changed network 

note that I've succeed in installing some others packages. 
If anyone have a clue ...

ameisso, 2017-05-02:
Still have my installation problem. 
Here is what i've tried so far :

- remove atom and .atom files 
- install older atom version (1.14) 
- tried the beta 
- tried to download from another connection
- unchecked "use proxy Settings When Calling APM"
- installing from command line (apm install platormio-ide) 

if anyone have a clue ...

ivankravets, 2017-05-02:
Do you use Proxy?

ameisso, 2017-05-02:
Nope, I don't have proxy's on any connection i've tried.

ameisso, 2017-05-07:
After testing it with the 2.0 beta 3 version (unsuccessfully),
I've reinstalled an older version by doing the following from my home folder : 

- `rm -r ~/.atom/packages/platformio-ide`
- download this version and copy the content of the folder inside `~/.atom/packages/platformio-ide`
- `cd ~/.atom/packages/platformio-ide`
- `npm install`
- restart atom 

Hope it helps someone 

Best

ivankravets, 2017-06-01:
Do you have problem with latest Atom 1.17 and PIO IDE Beta 4?

ameisso, 2017-06-02:
I've tried it and got pretty much the same issue 
"Virtualenv Download: Could not get proxy from APM: npm config failed: 1"

ivankravets, 2017-06-02:
Please open system terminal and type:
```
apm config get https-proxy
```

What do you see?

ameisso, 2017-06-02:
@ivankravets, I've tried it and this is what I get : 

`null`

regards

commit bbb5987e (2017-06-03): Fix issue when could not retrieve proxy // Resolve #313
closed 2017-06-03 by ivankravets