platformio/platformio-atom-ide #314: Install missed packages: linter-ui-default,intentions

svsummer, 2017-04-01:
Error: events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: ENOENT: no such file or directory, open '/home/stijn/.atom/packages/linter-ui-default/node_modules/sb-npm-path/.babelrc'
    at Error (native)

    at /home/stijn/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:86:27
    at onExit (/home/stijn/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/home/stijn/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/usr/share/atom/resources/app.asar/src/buffered-process.js:322:11)
    at /usr/share/atom/resources/app.asar/src/buffered-process.js:343:11
    at Socket.<anonymous> (/usr/share/atom/resources/app.asar/src/buffered-process.js:204:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

ivankravets, 2017-04-01:
Could you restart Atom, does it work now?

svsummer, 2017-04-03:
Yes, it works now. Sincere apologies for the overhead!

ivankravets, 2017-04-30:
Depends on https://github.com/platformio/platformio-atom-ide/issues/424

commit 9bb32114 (2017-05-01): Install dependencies one by one when failed to install all at once

Resolve #424, Resolve #285, Resolve #288, Resolve #292, Resolve #310,
Resolve #314, Resolve #315, Resolve #331, Resolve #334, Resolve #338,
Resolve #345, Resolve #358, Resolve #362, Resolve #380, Resolve #383,
Resolve #387, Resolve #394, Resolve #397, Resolve #401, Resolve #418,
Resolve #421, Resolve #332, Resolve #339, Resolve #342, Resolve #346,
Resolve #363, Resolve #381, Resolve #386, Resolve #395, Resolve #402
closed 2017-05-01 by ivankravets