platformio/platformio-atom-ide #316: InstallationManager

treaves, 2017-04-01:
Error: Error: events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: ENOENT: no such file or directory, open '/Users/treaves/.atom/packages/linter-ui-default/node_modules/minimist/.travis.yml'
    at Error (native)

    at AtomDependenciesStage.<anonymous> (/Users/treaves/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:164:13)
    at throw (<anonymous>)
    at step (/Users/treaves/.atom/packages/platformio-ide/lib/installer/stages/atom-dependencies.js:15:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)

ivankravets, 2017-04-01:
Could you restart Atom, does it work now?

treaves, 2017-04-01:
When I restarted, it re-downloaded the failed packages, and is now working.

closed 2017-04-01 by ivankravets